kbwbe/A2plus #5: manual constraints calculation

Turro75, 2018-07-03:
Hello again,

my laptop isn't exactly a gaming pc, could You please disable automatic constraints resolver? as the assembly increases in complexity the UI become quite unusable because every time I move/modify something the solver recalculate everything.

kbwbe, 2018-07-03:
Thank you for testing again.

 I never believed that someone tries to do some big assemblies with A2Plus as it is in experimental state. Great ! A real feedback !

I feared that your problem would be happening with some 20++ parts within assembly.

I am thinking about some solutions:

1) A preference page where you can disable this behavior and solve manually...

2) a new "freeze except selected part(s)" tool. Somehow working this way:
- you have an already big assembly, everything is correctly assembled, you imported a new part...
- you select the newly imported part and hit a button for "freeze except selected part(s) "
- the new part(s) will only be movable for the solver, the solver has few work. It will be fast then.
- After you unfreezed all objects and solving manual, the whole assembly will be recalculated. In this case you will have to wait some seconds. I think this could a good way to work with (very) big assemblies, also 
on notebooks.

3) The solver is still a prototype. Some things could be implemented to make it faster.
(this will take some time more for me)

Your opinion is greatly appreciated. Feel free to write a comment.

P.S. Did you have difficulties with constraints which are not working as expected ?
Your are the first one giving back detailed informations :-)  Thank you...

kbwbe, 2018-07-04:
Added button for enabling/disabling autoSolve of constraints.
Branch master updated.

closed 2018-07-04 by kbwbe