kbwbe/A2plus #19: Can't set color to an object

abetis, 2018-07-21:
Objects can't be set to a desired color, only random color works.
Looks like the opened dialog expects faces and not objects, so it doesn't allow to select the color.

kbwbe, 2018-07-21:
Did you import your parts ? For me (linux) setting color within assembly works. Usually an imported object consist only of faces (a compound)

abetis, 2018-07-22:
I'm on Linux as well, parts are imported properly.

Now I see that while in the color selection dialog, I have to select the faces one by one and than select the color... That way it works.
But all I want is set the needed color of the whole object. We need a button that will select all the faces for us instead of requirement to rotate the part to select all the faces. In case of a complex part it's insane.

kbwbe, 2018-10-02:
I tested setting of colors again. Everything is working for me. Please open new issue if there are still problems.

closed 2018-10-02 by kbwbe