kbwbe/A2plus #27: Undo constraint deletion places it in a wronk place in the tree

abetis, 2018-07-27:
When constraint is added, it appears one level down under the object that it's attached to.
When deleted and after that restored with Undo (Ctrl+Z) the constraint appears as a stand alone leaf in the tree and do not get back to its previous location.

kbwbe, 2018-07-29:
Added a new command "repair treeview", to be started manually at moment.
Do not know to call it automatically after undo has been done.
Perhaps you know how to do this...

abetis, 2018-07-29:
Have no clue. We should probably ask about it in the developers corner forum.

kbwbe, 2018-07-29:
I close this issue, as it is covered by your next one....

closed 2018-07-29 by kbwbe

abetis, 2018-07-30:
That one is more specific to constraint placement...