kbwbe/A2plus #44: A2plus does not support translation

WandererFan, 2018-08-13:
At some point, we will need to add 
def translate(context, text, disambig=None):
    return QtCore.QCoreApplication.translate(context, text, disambig)
and
QtCore.QT_TRANSLATE_NOOP()

to allow dialogs and tooltips to be displayed in user's preferred language.

luzpaz, 2019-10-15:
https://forum.freecadweb.org/viewtopic.php?f=10&t=36413

related: https://github.com/shaise/FreeCAD_SheetMetal/issues/99

luzpaz, 2020-03-22:
Some headway was made in the [forum thread](https://forum.freecadweb.org/viewtopic.php?f=10&t=36413&start=10#p375856). The solution (here's an example from the BIM WB:

```
        FreeCADGui.addLanguagePath(BimCommands.getLanguagePath())
        Log ('Loading BIM module... done\n')
        FreeCADGui.updateLocale() # refresh the language according to current locale.
```

luzpaz, 2021-02-18:
@yorikvanhavre started a wiki page https://wiki.freecadweb.org/Translating_an_external_workbench

luzpaz, 2021-02-18:
Started working on this at https://github.com/kbwbe/A2plus/pull/416  
Feel free to help, my branch is at https://github.com/luzpaz/A2plus/tree/translation

kbwbe, 2021-02-28:
Preparations are done within #421

luzpaz, 2021-02-28:
What's left to do?

kbwbe, 2021-02-28:
The translation does not work.

luzpaz, 2022-08-28:
@kbwbe is this ticket obsolete now ?

kbwbe, 2022-08-29:
@luzpaz: I think @PushKK  did the job.

luzpaz, 2022-08-29:
The this ticket is obsolete

closed 2022-08-29 by kbwbe