kbwbe/A2plus #88: A2plus Feature Wishlist

ManuelKrause, 2018-09-06:
As the A2plus assembly workbench for FreeCAD gets more attention over time, users may have wishes for e.g. added features, added constraints to apply and convenience enhancements etc.

This thread should give room for adding wishes. 

First goals are still:
* improving accuracy, code cleanup, testing, NOT introducing BUGS
* implementing new solver additions, improving importer regarding updating of parts

ManuelKrause, 2018-09-06:
First named feature wishes came from @Turru75 in the freecad forum:

1) as assembly2 does, the multiple constraints for screws, bolts,... very useful, I guess that hamish's code is nearly ready to be integrated
2) be able to enable/disable a constraint without the need to delete and recreate it.
3) constraint to system axes
4) symmetric constraint
5) make symmetric 2 faces of a part to 2 faces of another part (I'll provide an example at the right moment), I found it very useful in the past, in the end it is a plane constraint where the offset is defined by the distances of faces.
6) direct stp importing, and hopefully a command to make all objs in the current file as a2p parts
7) BOM to spreadsheet

and as last...
8) a command that temporary change color of objects according to its DOF, i.e. red fully constrained, amber partially constrained, green free to move, blue overconstrained or something like that

ManuelKrause, 2018-09-06:
@Koemi: added his wish in the forum: 
When adding a plane constraint a screen pops up asking for a distance (default=0); when necessary you can directly type in the value; if not necessary simply just click next to the popup-screen to make it disappear again. SolidWorks does this trick also.

luzpaz, 2018-09-06:
@ManuelKrause you can use checkboxes
example:  
- [ ] BOM to spreadsheet
- [x] symmetric contraints

ManuelKrause, 2018-09-06:
@luzpaz:
Let's keep on only collecting ideas on here at first. Later @kbwbe decides what to do next. It's his project and only he decides the appropriate development pace.

WandererFan, 2018-09-09:
Conversion routines from A2, v0.16 and earlier versions of A2p.   Backwards compatibility was one of the big complaints in the v0.17 release.

rkoeman, 2018-09-14:
Move any part inside the assembly directly by LMB-click + hold + swipe. Movement possible according to it's constraints. So if a part has DOF it can move (rotating, sliding etc.). Without DOF it stays. The 'Move part'-button necessity vanishes (I think).

slhh, 2018-09-19:
Try to reuse the contraint icons of the sketcher, and use less, but more universal contraint icons where applicable.
For a example:
A 'coincident' icon can be used to generate the following constraints based on the type of selection:

- point to point coincidence
- line to line coincidence
- plane to plane coincidence

Even the following contrains might share the same icon, because it's still some kind of partial coincidence:
- point on line
- point on plane
- line on plane
- line on cylinder

A 'parallel' icon can be used to generate the following constraints:
- line parallel to line
- line parallel to plane
- line parallel to cylinder
- plane parallel to plane

polemidis, 2018-10-28:
- A constraint that I regularly miss is a circle between a slot!

- I liked the new feature recently added to the assembly2, when a new constraint was created, a pop up 
screen asked for some extra text. So the user could easily add a "z" or "y" or some other identification to the text string of the constraint. Made it easier to identify them later. Of course a graphical approach could be used later, but the text solution should be easily adopted for now

silopolis, 2018-10-28:
Le dim. 28 oct. 2018 à 02:59, Manolis Karamousas <notifications@github.com>
a écrit :

> -
>
>    I liked the new feature recently added to the assembly2, when a new
>    constraint was created, a pop up
>    screen asked for some extra text. So the user could easily add a "z"
>    or "y" or [...]
Seconded, I'm using it for all the constraints I create

kbwbe, 2018-11-04:
>     * A constraint that I regularly miss is a circle between a slot!
This has been done within version V0.1.5.1 
Use pointOnPlane with center of circle as point-selection. Now it is possible to use an offset.

silopolis, 2018-11-21:
Hi,
I'm badly missing the "fork" feature. Used to use it constantly in a kind of evolutive design habit.

f4grx, 2019-01-16:
Hello,
What I miss is the ability to use a circle cut perpendiculary to a cylinder for "same axis" constraints, eg I have no way to constrain this tube in this hole:
[image]

Thanks for your awesome plugin.

Turro75, 2019-01-16:
Hello,

do You mean this?

[image: image.png]

walterwissmann, 2019-02-12:
We have an 'open subassembly'-button:
[image]

It would be very helpfull to have also a 'save and close'-button
[image]

to navigate fast trough subassemblies

f4grx, 2019-02-13:
@Turro75 Sorry for the late reply. I cant see your image :(

Turro75, 2019-02-13:
@f4grx

> @Turro75 Sorry for the late reply. I cant see your image :(

[image]

f4grx, 2019-02-14:
Turro: exactly!

So how do you do that, I cant find an axis for the intersecting hole.

Turro75, 2019-02-14:
Simply I selected the green parts and then set an axis constraints

Regards

walterwissmann, 2019-02-17:
- Dimensions Draft-WB comes not through subassemblies
- Pictures comes not through subassemblies

These two techniques are day by day work pieces. They should come.

Walter

polemidis, 2019-02-18:
Feature request plz:
The conversion of fasteners to A2+ parts should allow multiple items selection :)

easyw, 2019-02-24:
@polemidis @kbwbe 
> Feature request plz:
The conversion of fasteners to A2+ parts should allow multiple items selection :)

please have a look at my latest [PR](https://github.com/kbwbe/A2plus/pull/238) ... it should address this feature request

walterwissmann, 2019-03-14:
Hi,
i have found an issue with transparency. I am not sure it comes from A2Plus... So, i will send the files for test it to you. But it is an assembly with around 30 Files in around 10 folders.

Now my Idea is A2plus could collect them in a ZIP-File...
This is also practical to send huge assemblies to other developers ;)

VG
Walter

ManuelKrause, 2019-04-01:
@walterwissmann 
The transparency issue is known and present since months. This particular thread is not the place to complain about it. 
And by the way, you don't need to send huge assemblies related to your original post. If current FC code goes wrong it also occurrs with smaller assemblies! (2nd level with one transparent part vs. one fully colored)
I'm really trying my best to test other color-handling variants, but haven't found enough time for all so far.

Manuel

walterwissmann, 2019-04-01:
Ok, 
thank you, Manuel.

BR
Walter

Perlovka, 2019-04-11:
Is it possible to highlight selected constraints?
E.g. when you select constraint in parts tree, it's subelements becomes highlighted in main window.

ManuelKrause, 2019-04-16:
> Hi, is it possible to highlight selected constraints?
> E.g. when you select constraint in parts tree, it's subelements becomes highlighted in main window.
> Thanks.

This is already effectively working with A2plus:
Just select the constraint and then click the (red-lighted) Lense from the A2plus WB, then you get a highlighted view of your constraint.

ManuelKrause, 2019-04-16:
Can there be a contraint for a circle <-> circle tangency?
In the following meaning:
I want to attach one cylinder's outer shape to one other cylinder's inner shape.

Perlovka, 2019-04-16:
@ManuelKrause thanks! I missed this button )

kbwbe, 2019-04-16:
> Can there be a constraint for a circle <-> circle tangency?
> In the following meaning:
> I want to attach one cylinder's outer shape to one other cylinder's inner shape.
> This may be also interesting for some gear assemblies, attaching one radial face to another.

Hi Manuel,
there are several useful constraints still missing. Your case could be managed by an axisOffset constraint.

Others useful constraints could be:
- pointPointDistance
- pointLineDistance
- angleBetweenLines
- angleLineToPlane
- etc ...

Each new constraint causes a lot of work, especially the tests and the debugging in 3D. I hope i can add some few new constraints within next time. (this could be several month)

ManuelKrause, 2019-04-17:
Hi @Turro75 :
I've had taken some of your feature wishes from the FC forum to this Wishlist in last September:
One of them, cited below, seems to be solved for me by A2plus' ability to attach colored labels for the DOF of parts.

> and as last...
> 8) a command that temporary change color of objects according to its DOF, i.e. red fully constrained, amber partially constrained, green free to move, blue overconstrained or something like that

Does it fulfill your needs? Then I can strike it in the top posting's list.

Turro75, 2019-04-19:
Hi Manuel,

Yes it is

m0n5t3r, 2019-04-23:
Not sure if this is the right place or it should be filed as a bug, but:

The ability to import multiple parts from the same fcstd file would be nice; any moderately complicated thing will have parts that depend on other parts, so the current working workflow is to export all parts as stp and import into a2plus for assembly, and then re-export everything when things change...

What I expected to happen when I tried to import a fcstd file with multiple objects was something along the lines of "here's a list of bodies / parts from the file, which ones do you want to import?"; I'm ok with requiring bodies to be placed in part containers for this, as a means of signaling "this is a thing you should import as one object".

ManuelKrause, 2019-04-24:
@m0n5t3r :
I actually don't get your point. Can you please explain your use case a little more?
Normally I'd set objects of a part .fcstd as invisible, if I don't want to work with them for assembly purposes. The A2plus assembly WB would only work with the visible objects then. Upon changes the A2plus reload button would recover the scene (and constraints) if all shown objects' number remains the same. 
You could also divide the current multiple-parts .fcstd into multiple single object's .fcstd s and assemble them with A2plus, but that's the additional work you want to avoid IIUC.
Maybe better, you just ask the same question in the Forum https://forum.freecadweb.org/viewtopic.php?f=20&t=29207&start=1600. There are many helpful and experienced people.

Best regards

m0n5t3r, 2019-04-24:
say I have a file with multiple objects (in my case, quadcopter frame - there's the base frame, 4 legs, a lower plate, 4 small reinforcement parts); they need to be in the same file because all dimensions / shapes depend on the main frame's dimensions (shape binders)

Right now, I need to export each of them to a separate file and then import those into the assembly; it would be nice if it was possible to import all existing objects from the file as separate parts (because it imports everything as a single part now, even if I have a single thing visible)

ManuelKrause, 2019-04-25:
@m0n5t3r:
 Ah yes, thank you for your explanation! Now your proposal makes sense. 
I haven't had knowledge of or used this way of construction, so far, thanks for letting me know. :)
Let's wait and see what Klaus, @kbwbe, the A2plus workbench maintainer, can say regarding your idea. Maybe he can point you to another more simple way to achieve your goal with current A2plus' infrastructure.
Your proposal, from my first thoughts about, would make it necessary to implement a list of imported (3D) objects per import file within the assembly file, where the entries are pointing to the different (3D) objects in the subordinate file.

kbwbe, 2019-04-26:
@m0n5t3r
Why do you export all parts separatly to a file and import them separatly to A2plus ? If you have done an important sub assembly within e.g. Part-Design, you could import this complete document to A2plus in one step. There are draw backs regarding parts list. But for first design of whole concept, a simplified workflow is possible by this way. If the design has been completed, PD docs could be split up for separate production documents.

zmp2000, 2019-05-03:
@kbwbe What do you mean by "important sub assembly within e.g. Part-Design"
I have a document with 2 Distinct parts (and one different body for each) when I import the file , They all came as one part! And so I can not constraint anything, Am I missing something?

easyw, 2019-05-04:
@m0n5t3r @zmp2000 
please have a look at here https://github.com/easyw/Manipulator/issues/20
there is a mixed solution using A2p and Arch external references...
@kbwbe External references, as used by Arch could be useful also for A2p WB when a user needs to have parts in the same file because all dimensions / shapes depend on the main frame's dimensions (shape binders)
[Arch external reference forum topic](https://forum.freecadweb.org/viewtopic.php?t=32998#p278706)

m0n5t3r, 2019-05-04:
external references seem to work, indeed, but need to be converted to a2plus objects: https://forum.freecadweb.org/download/file.php?id=82820 (from https://forum.freecadweb.org/download/file.php?id=82820)

easyw, 2019-05-04:
@m0n5t3r
infact, the idea would be to add to A2p a field to chose what to import like in Arch Ext Ref.
Then when importing it would be possible to import all visible objs or a single Part feature like in Arch.
@kbwbe what do you think?

kbwbe, 2019-05-04:
@easyw 
yes, adding such field makes sense. As A2p objects are not totally different from Arch Ext Ref in basic idea. I will have a look at yorik's source code /freecad/Mod/Arch/ArchReference.py i guess.

zmp2000, 2019-05-04:
@easyw  I don't follow. Sorry if it´s a dumb question (I´m a newbie) but when I import using **external references** (E, X shortcut), the parts imported don't have a property "fixed position" and so when I try to create any constraints I got a waning (No parts with fix position) and constraints don't work!

As @m0n5t3r stated, it only works if we convert it to A2Plus object, but than we loose the ability to update the object!

easyw, 2019-05-04:
> As @m0n5t3r stated, it only works if we convert it to A2Plus object,

true
> but than we loose the ability to update the object!

ATM you can
1) create a file using Arch Ext Ref (i.e. arch-ext-ref-1.FCStd)
2) create a file in A2p  (A2p-asm.FCStd) and import the previous file (arch-ext-ref-1.FCStd)
3) when you want to update the A2p-asm.FCStd you need to update and save your Arch Ref file (arch-ext-ref-1.FCStd) and then update the top level file (A2p-asm.FCStd)

BTW I think @kbwbe is going to implement this option inside A2p WB, avoiding this workarund :smile:

kbwbe, 2019-05-05:
@easyw , @m0n5t3r, @zmp2000 
Hi guy's,
please checkout the devel branch and test. There is a second new import command, which imports selectable bodies from a multibody document.
This is the button:

[image]

easyw, 2019-05-05:
@kbwbe 
it seems to working fine... it is a nice addition! :smile_cat: 
I think you missed to add 
@ [InitGui.py#L155](https://github.com/kbwbe/A2plus/blob/4b270714a5e514a48b45391b96cc588f8205a641/InitGui.py#L155)
`                'a2p_ImportShapeReferenceCommand',`
otherwise you get the command enabled only if recursive update is enabled.

m0n5t3r, 2019-05-05:
confirming it works, I had to dig through code to figure out it needs recursive update :)

zmp2000, 2019-05-06:
@kbwbe It works like charm after adding the import 'a2p_ImportShapeReferenceCommand'

kbwbe, 2019-05-06:
I updated the devel branch. The new button is now available without need to activate recursive updates. Also a first small bug is fixed (function getIcon() of importedParts got struggled)

ManuelKrause, 2019-05-06:
@easyw , @m0n5t3r, @zmp2000, @kbwbe
Hi people,
yet another nice example for successful collaboration based on the "open-source" principles. Good ideas, good hints and a good implementation.
Thank you all for your invested time and work.
I've just tested it for fun with a true A2plus assembly (although I know there are better means, e.g. to avoid future confusion) and it works upto one level below (where the assembled part names are shown).

ManuelKrause, 2019-05-06:
Hi again,
thanks for using this thread for helping to improve A2plus.

When reviewing the proposals list i get to some limits.
Currently I have four categories:
(a) I dont understand it, due to language misunderstandings.
(aa) I don't understand it without more detailed use-case description.
(aaa) All info clear, it's developer's choice to pick it up or not.
(aaaa) Not sure, if applicable within FreeCAD so far.
I'd focus on the first two categories first:

Please help me to improve your proposal by adding more information:
(a) language:
2.1 "as assembly2 does, the multiple constraints for screws, bolts,..." @Turro75
Please clarify what you mean with "the multiple constraints".
8.1 "fork" feature (?evolutive design habit?) @silopolis
I completely don't understand this. Please carify!
(aa) use case/ more info needed:
2.8 BOM to spreadsheet @Turro75
What do you expect happenening, in more detailed description?
6.1 Use more universal Icons where applicable @slhh
Do you want more and different Tools grouping?
7.2 Popup for Constraint's name/text editing: @polemidis
I can edit the Constraint's name in the tree view. Isn't that sufficient for you?

@kbwbe:
Regarding the following:
3.1 distance value entry popup for plane constraints and skip:
Wouldn't it indeed be a good idea to add a skip button to the planeCoincident command (and maybe related), to give the ability to revert the movement after having hit "Solve" before reverting it? ATM parts remain in the solved state anyway. This is not always the wanted position.
7.2 regarding "Popup for Constraint's name/text editing:" testing I've noticed this:
Maybe a BUG?: After editing the first constraint's name in TreeView the counterpart's mirror constraint doesn't take it up.

ManuelKrause, 2019-05-09:
@kbwbe:
for 3.1 "distance value entry popup for plane constraints and skip", the planes' distance part is fulfilled, of course. 
Maybe a combined revert=skip usage meaning would be useful: To move back parts to the original positions in that case of constraint cancelling..

Turro75, 2019-05-09:
> Hi again,
> thanks for using this thread for helping to improve A2plus.
> 
> When reviewing the proposals list i get to some limits.
> Currently I have four categories:
> (a) I dont understand it, due to language misunderstandings.
> (aa) I don't understand it without more detailed use-case description.
> (aaa) All info clear, it's developer's choice to pick it up or not.
> (aaaa) Not sure, if applicable within FreeCAD so far.
> I'd focus on the first two categories first:
> 
> Please help me to improve your proposal by adding more information:
> (a) language:
> 2.1 "as assembly2 does, the multiple constraints for screws, bolts,..." @Turro75
> Please clarify what you mean with "the multiple constraints".
> 8.1 "fork" feature (?evolutive design habit?) @silopolis
> I completely don't understand this. Please clarify!
> (aa) use case/ more info needed:
> 2.8 BOM to spreadsheet @Turro75
> What do you expect happenening, in more detailed description?
> 6.1 Use more universal Icons where applicable @slhh
> Do you want more and different Tools grouping?
> 7.2 Popup for Constraint's name/text editing: @polemidis
> I can edit the Constraint's name in the tree view. Isn't that sufficient for you?
> 
> @kbwbe:
> Regarding the following:
> 3.1 distance value entry popup for plane constraints and skip:
> Wouldn't it indeed be a good idea to add a skip button to the planeCoincident command (and maybe related), to give the ability to revert the movement after having hit "Solve" before reverting it? ATM parts remain in the solved state anyway. This is not always the wanted position.
> 7.2 regarding "Popup for Constraint's name/text editing:" testing I've noticed this:
> Maybe a BUG?: After editing the first constraint's name in TreeView the counterpart's mirror constraint doesn't take it up.

Hi Manuel,

[image]

in assembly2 there was the bolts feature which allowed to automatically insert screws in holes simply selecting a screw then a copy of it is created directly on any circle edge selected with creation of constraints. If I'm not wrong it worked on any asm2 object not only screws.
unfortunately I have no more asm2 running on my pc so I cannot make a gif to show You how it worked.

Regarding the BOM to spreadsheet, the target was that a freecad spreadsheet is created containing the BOM. Then the BOM can be easily handled even externally from Freecad.
I think Part list fullfill these part of the wishlist.

Best Regards

Valerio

kbwbe, 2019-05-09:
> @kbwbe:
> for 3.1 "distance value entry popup for plane constraints and skip", the planes' distance part is fulfilled, of course.
> Maybe a combined revert=skip usage meaning would be useful: To move back parts to the original positions in that case of constraint cancelling..

Hi  @ManuelKrause,
there are 2 cases if you want to revert a constraint.
No 1: A constraint has been set and the parts have been moved.
No 2: A constraint has been set and the parts have not been moved because of a not solvable system

The problematic case is only No 1. But after deleting the constraint within the constraint properties dialog it is possible to use system undo function to get the parts back to original position. Last time i did a big assembly of a complete machine. (more than 300 parts over all sub assemblies). There was need to use the undo function about 2-3 times. IMHO the dialog is sufficient and in special cases, there is still the undo function.

Best regards, Klaus

kbwbe, 2019-05-09:
I add a wish to my own WB...
Most thing i missed during real world work was the absence of patterns (linear or polar) with correct representation in parts-list.
Has anyone an idea how to gain this ?

ManuelKrause, 2019-05-10:
> Hi Manuel,
> 
> [image]
> in assembly2 there was the bolts feature which allowed to automatically insert screws in holes simply selecting a screw then a copy of it is created directly on any circle edge selected with creation of constraints. If I'm not wrong it worked on any asm2 object not only screws.
> unfortunately I have no more asm2 running on my pc so I cannot make a gif to show You how it worked.
> [...]
> 
> Best Regards
> 
> Valerio

Hi @turro75, Valerio,

thank you very much for refining your proposal. It's now a long time that I am not able to get an asm2 working here and your description now gives me an idea what you mean. 
Would the words "Multiply imported parts (like screws, bolts, etc.) to selected circular edges" better describe it? Or even "Import multiple parts (like screws, bolts, etc.) to selected circular edges"? Please let me know, what wording you like more and feel free to give an own alternative (so I'd change the wishlist entry accordingly).

TIA, Manuel

ManuelKrause, 2019-05-10:
> [...] it is possible to use system undo function to get the parts back to original position. Last time i did a big assembly of a complete machine. (more than 300 parts over all sub assemblies). [...]

Hi Klaus, @kbwbe,
yeah, thank you for your input. I don't know why I always forgot to use this basic FreeCAD Undo function. When having this in mind, this item is fulfilled IMO.

Do you have something for this
> 7.2 regarding "Popup for Constraint's name/text editing:" testing I've noticed this:
> Maybe a BUG?: After editing the first constraint's name in TreeView the counterpart's mirror
> constraint doesn't take it up.

? (Or is this only happening to me? I haven't updated FreeCAD for several weeks now.)

Best regards,
Manuel

ManuelKrause, 2019-05-10:
> I add a wish to my own WB...
> Most thing i missed during real world work was the absence of patterns (linear or polar) with correct representation in parts-list.
> Has anyone an idea how to gain this ?

Hi again, @kbwbe,
just to say it again, I'm not safe with Assembly2 code anymore. But hasn't there been some linear/ polar pattern code in there, around about a year ago? Maybe, if it indeed existed, you can re-use that code somehow. I have several asm2 states archived on disk, but am clueless ATM for what keywords to search for.
For the parts-list representation, this is beyond my knowledge. Maybe you just only need to adjust your existing "data-collection" functions, or even not, you'd then collect the pattern related files/ parts, once they are in the assemby, just like any other assembled part.

henkjannl, 2019-06-20:
I'd be very grateful with an interference check function. If I'm correct that worked fine in Assembly2.

jophut, 2019-08-10:
I'd like to use a datum plane in a coincident relation.

Currently, for imported parts only the 'tip' is visible, all other features are hidden.

From my SolidWorks experiences I sometimes find it very useful to use features from sketches or reference planes/axis/points for relations.

E.g:
- I have two parts P and Q, both with a single body (e.g. a cube) and an empty assembly A.
- In part P I also have a datum plane at some distance from the solid body.
- In the assembly I first import P and fix it at the origin.
- Then I import Q and want some surface of the body to be coincident with a datum plane from part P.

This way I have P and Q some distance apart. The nice thing is that it's defined by the datum plane in part P.

Is it be possible to use the Python console to accomplish this?

meprakarsa, 2022-03-23:
Requesting: **Animation menu:**
1. **Select** A2plus part to animate
2. Select animation method:
    - **rotate** cw/ccw **from** start_angle **to** end_angle
    - **shift** x/y/z
    - **shift+rotate follow a sketch**

3. **start**, **pause**, **end** animation button (every animation step is of course followed by "solve A2plus assembly"

thank you from noob here