kbwbe/A2plus #101: imported parts should support relative path

ouinouin, 2018-09-23:
Hi, 
In order to contribute to an assembly project with several people, a good point would be to be able to support relative path (./mypart1.fcstd instead of /home/joe/myproject/mypart1.fcstd ) in the source file field in order to be able to share a zip containing all the files with other people or to be able to put the files in a versioning system.
I saw the option to put everything in the same folder in the preferences menu but it seems to be different to what is expected from a relative path.

silopolis, 2018-09-24:
Can't second this one enough !
Default should be configurable, and path property should accept both relative and absolute path.
Absolute path is an incredible limitation in projects and parts libraries files and directories management.
Icing on the cake would be simple functions for path and files management: convert to absolute, convert to relative, move/copy files to project dir...

kbwbe, 2018-09-25:
Hi,
i will enable relative path with version v0.2. Please use until this the project folder option from preferences. Using this, path is ignored and the filename is looked up within content of projectfolder, even in subfolders.
This behavior is alike Autodesk Inventor.

kbwbe, 2018-09-28:
Devel branch is already working with relative pathes. A tool to convert existing absolute pathes to relative ones can be found within a2p menu. Master branch will be updated after testing.

closed 2018-09-28 by kbwbe