kbwbe/A2plus #108: Part/Mirror does not gets imported

polemidis, 2018-10-27:
Hello guys, thats maybe a quick fix. 
When I import a part where a body is mirrored with the Part/Mirror,  the mirror bodyt is imported but not the original body.

Is this intentional? I have a turnaround by making a fussion between the 2 bodies, but why adding more steps?

Thank you for the great job! 
P.S. Do you want to put new issues that I found here or in the freecad forum?

kbwbe, 2018-10-27:
Thanks for the hint ! This is a bug. Reporting bugs here on Github is sufficient. I look regularly to issues.

polemidis, 2018-10-28:
Perfect man! As I am transitioning into A2+ I will post here whatever I found. Thank you!

kbwbe, 2018-10-28:
Fixed with PR #110 
V0.1.4.1 is working now for this.

closed 2018-10-28 by kbwbe