kbwbe/A2plus #109: Does not constrain non-imported parts.

polemidis, 2018-10-28:
I mean if I create 2 bodies on the same file, I cannot constraint them. Is it intentional? My apologies if that is the case. I found out that I can use the assembly2 constraints specifically for this case,  but I am worried that it will complicate things or cause some problems later.

polemidis, 2018-10-28:
No, it does not work. Even though assemby2 get the parts in place, when I import that file to an assembly the position of the parts are not constrained.
[saw system shaft attachment angle2.zip](https://github.com/kbwbe/A2plus/files/2522369/saw.system.shaft.attachment.angle2.zip)
This is the file. Its an .fcstd file, I changed the extension so I can upload it. If you open the file it looks that the cylinder is lined up with the hole, but if you import it with A2+ the cylinder is out of place. 

Hopefully that helps.

polemidis, 2018-10-29:
Yes indeed! Something is very weird!
Ok thanx for letting me know. I didn't know the Manipulator WB. :) :)

kbwbe, 2018-10-31:
This is problem is not caused by A2p.

closed 2018-10-31 by kbwbe