kbwbe/A2plus #113: on V0.1.5 the Partlist crashes Freecad-daily 0.18

polemidis, 2018-11-03:
The partlist crashes on these tested scenarios
a) Either-way I choose to do a recursive search of sub-assemblies or not
b) When I press the Partlist button after I open my assembly which is [here](https://gitlab.com/go_commons/firewood-processor) 
I tried it either with the "0-Master-CAD.fcstd"  and with a smaller "saw bar assembly.fcstd" crashes on both

Hopefully that helps on debugging :) :) 

Thank you guys!

kbwbe, 2018-11-04:
Please try new release V0.1.5.1. This worked for me with your files. 

Your files seem to be done with older Versions of A2p. The property "subassemblyImport" is missing within the A2p objects. I fixed this, but for older files A2p is not able then to iterate over subassemblies, to get a whole collection of all parts. You have to do a separate partlist for each subassembly. Sorry !

polemidis, 2018-11-07:
Still crashes down right away. Let me know if I can help debugging this

kbwbe, 2018-11-10:
Hi polemidis,
perhaps it is better to do some more conversation in my forum thread. Please post there some more information, as e.g. the console output of a crash.

polemidis, 2018-11-12:
absolutely. Do you mean the freecad forum?

polemidis, 2018-11-12:
Here is the console output.
`Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x41100) [0x7fa20bed7100]
#1  /lib/x86_64-linux-gnu/libexpat.so.1(XML_SetHashSalt+0x1b) [0x7fa204ac895b]
#2  /usr/lib/python2.7/lib-dynload/_elementtree.x86_64-linux-gnu.so(+0x8660) [0x7fa1d80a6660]
#3  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x9327) [0x7fa20d525b47]
#4  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x8e5f) [0x7fa20d52567f]
#5  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x8e5f) [0x7fa20d52567f]
#6  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x732) [0x7fa20d51c0c2]
#7  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalFrameEx+0x6d33) [0x7fa20d523553]
#8  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_EvalCodeEx+0x732) [0x7fa20d51c0c2]
#9  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x195a1b) [0x7fa20d5a2a1b]
#10  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x53) [0x7fa20d5cce93]
#11  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(+0x1acb8c) [0x7fa20d5b9b8c]
#12  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyObject_Call+0x53) [0x7fa20d5cce93]
#13  /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0(PyEval_CallObjectWithKeywords+0x47) [0x7fa20d51b757]
#14  /usr/lib/freecad-daily/lib/libFreeCADBase.so(PP_Run_Method+0xe7) [0x7fa20d8fd0f7]
#15  0x7fa20d886aa9 in Base::InterpreterSingleton::runMethodVoid(_object*, char const*) from /usr/lib/freecad-daily/lib/libFreeCADBase.so+0x29
#16  0x7fa20e1320a8 in Gui::PythonCommand::activated(int) from /usr/lib/freecad-daily/lib/libFreeCADGui.so+0x88
#17  0x7fa20e135274 in Gui::Command::invoke(int) from /usr/lib/freecad-daily/lib/libFreeCADGui.so+0x54
#18  0x7fa20c3c4f70 in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x4d0
#19  0x7fa20c8d6dc2 in QAction::triggered(bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x42
#20  0x7fa20c8d8113 in QAction::activate(QAction::ActionEvent) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x83
#21  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x5a0fa3) [0x7fa20ccb8fa3]
#22  0x7fa20ccb90f4 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x74
#23  0x7fa20cd7f0aa in QToolButton::mouseReleaseEvent(QMouseEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0xa
#24  0x7fa20c933930 in QWidget::event(QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x2a0
#25  0x7fa20c8dd03c in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x8c
#26  0x7fa20c8e4136 in QApplication::notify(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x476
#27  0x7fa20e10ade8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /usr/lib/freecad-daily/lib/libFreeCADGui.so+0x88
#28  0x7fa20c3b08ad in QCoreApplication::notifyInternal(QObject*, QEvent*) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x8d
#29  0x7fa20c8e373d in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x14d
#30  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x249452) [0x7fa20c961452]
#31  0x7fa20c960ce3 in QApplication::x11ProcessEvent(_XEvent*) from /usr/lib/x86_64-linux-gnu/libQtGui.so.4+0x16b3
#32  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x272592) [0x7fa20c98a592]
#33  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2ae) [0x7fa206f5dc3e]
#34  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4ded8) [0x7fa206f5ded8]
#35  /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c) [0x7fa206f5df6c]
#36  0x7fa20c3e120e in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x7e
#37  /usr/lib/x86_64-linux-gnu/libQtGui.so.4(+0x272666) [0x7fa20c98a666]
#38  0x7fa20c3af12f in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x3f
#39  0x7fa20c3af495 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x1b5
#40  0x7fa20c3b5459 in QCoreApplication::exec() from /usr/lib/x86_64-linux-gnu/libQtCore.so.4+0x99
#41  0x7fa20e0a9832 in Gui::Application::runApplication() from /usr/lib/freecad-daily/lib/libFreeCADGui.so+0x1792
#42  freecad-daily(main+0x6e8) [0x563e198f6ce8]
#43  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xeb) [0x7fa20beba09b]
#44  freecad-daily(_start+0x2a) [0x563e198f6fda]
`
Hope that helps. Feel freee to ask anything

kbwbe, 2018-11-12:
OK ! This is a complete FreeCAD Crash. It is usually not caused by any Python program. Perhaps the elementree lib. It is possible to post a small assembly with 2 parts (including the source files), which crashes ?

polemidis, 2018-11-14:
this is my repository [(https://gitlab.com/go_commons/firewood-processor)]
Its big but may its faster ;)  The cad files are in the "01_Hardware/CAD" folder. The "0-Master CAD.fcstd" crashes. I also checked a simpler assembly, the "Frame Module.fcstd" and the "Loader Module.fcstd" which also crash. Can you see if these crash on you plz? I will also try the 0.17 as soon as I can get it for ubuntu 18.10

kbwbe, 2018-11-15:
[image]
Hi polemidis,
for me, it works and i get no crash. I downloaded the firewood-processor repo. Working with relative pathes, i could open every file. for 3 files i did a partInfo spreadsheet. The image shows the list i get. It even worked for subassemblies, where possible. (clamp plate of cylinder base_001). The included pivot point plate is shown.

i tested with:
OS: Linux Mint 18.3 Sylvia
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15190 (Git)
Build type: Release
Branch: master
Hash: c18785fdb31e378500440de23dcf648a2c79dc5f
Python version: 2.7.12
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

and.. recent A2plus

No crash, so difficult to say what's happening on your system ?!

polemidis, 2018-11-16:
ok, I will investigate it a little more. 
My version is this 
OS: Ubuntu 18.10
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15215 (Git)
Build type: Release
Branch: master
Hash: 54799368944d408a09fcdde41f89f20d282a9b8a
Python version: 2.7.15+
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
and the latest A2+

kbwbe, 2018-11-20:
Hi Polemidis,
another FC user has problems with Assembly2, which has to do with libexpat, which is referred in your error trace. See this thread in FC forum: [https://forum.freecadweb.org/viewtopic.php?f=20&t=32338](url)

I fear, elementtree (for reading xml) seems to use libexpat in a buggy version.

polemidis, 2018-11-29:
Thank you, I will check it!

polemidis, 2018-11-29:
I downgraded the libexpat* packages,and now I am getting this, but at least the Freecad does not crash:
Anyway its ok, it looks like the problem is the new version of that library.

Running the Python command 'a2p_CreatePartlist' failed:
Traceback (most recent call last):
  File "/home/pomo/.FreeCAD/Mod/A2plus/a2p_bom.py", line 167, in Activated
    recursive=subAssyRecursion
  File "/home/pomo/.FreeCAD/Mod/A2plus/a2p_bom.py", line 91, in createPartList
    docReader2.openDocument(linkedSource)
  File "/home/pomo/.FreeCAD/Mod/A2plus/a2p_fcdocumentreader.py", line 187, in openDocument
    f = zipfile.ZipFile(fileName,'r')
  File "/usr/lib/python2.7/zipfile.py", line 779, in __init__
    self.fp = open(file, modeDict[mode])

[Errno 2] No such file or directory: 'converted'

kbwbe, 2018-11-29:
Hi Polemidis,
the recent error dump now shows an error of A2plus. I will check.

kbwbe, 2018-12-14:
Bug regarding converted parts in Partlist fixed within devel branch. 2018-12-14
Release will follow soon.

closed 2018-12-14 by kbwbe