kbwbe/A2plus #128: A2p doens't update recursively sub assemblies

easyw, 2018-12-19:
If in an assembly there are sub-assemblies, and a sub-part of a sub-assembly is changed, the user needs to update manually each sub-assembly that has this sub-part changed ... 
the desired behavior would be to have an automatic update of sub-assemblies, when the A2p update button is clicked...

kbwbe, 2018-12-19:
Hi easyw,
an old version (before going public) did updating recursively. I did not activate this on A2plus at moment, because there is still a risk that subassemblies "explode" due to topological naming issue. Therefore i prefered the "supervised" mode doing this step by step manually. I consider to make an option in preferences, so user can select his desired update mode.

easyw, 2018-12-19:
> I consider to make an option in preferences, so user can select his desired update mode.

Yes please add this option... this would be very useful :smile:
I have used A2p in a simpler way to obtain an assembly without adding constraints. In this way the topological naming issues are not present at all 😉 
I just placed the Assembly parts to the correct position using the Aligner (from Manipulator WB) and the rest is done by A2p WB 🥇 
[A2p using absolute positioning](https://forum.freecadweb.org/viewtopic.php?f=20&t=32843&p=275131#p275131)

kbwbe, 2018-12-19:
I will implement it within this or next release latest.

kbwbe, 2019-01-14:
Within devel branch it is ready for testing now.

closed 2019-01-14 by easyw