kbwbe/A2plus #139: Left over selected parts after "Update Parts"

polemidis, 2018-12-29:
There is an issue after the last update(s) that some parts stay selected without me noticing and I delete them by mistake. Even in the inactive document!. Sometimes I catch it because it throws en error of "are you sure you want to delete this because that will brake etc" and I am wondering why this showed up since what I want to select is has not dependencies. Maybe its a Freecad problem, i do not know yet. In the beginning I though it was my impression but I am getting it more and more. I have not being able to replicate it yet in order to pin it down, but I will add more specific info soon. I posted just in case you may suspect something already.

Also another thing that may be easy to improve is this. If i have a "Circular Edge Connection" [(did you mean to write Circular Edge Constraint? :)] and I update something minor in an imported part, everything stays solid after the "Update Parts Imported in the Assembly" . But if I use the "Axial Constraint" instead, every super tiny irrelevant change on the updated part will brake everything! :) :)

Feel free to close this issue, I will post a new one when I will have a solid reproducable bug. 
Thank you man, you have done GREAT GREAT job already with this!!!

polemidis, 2018-12-29:
Update, this is screenshot. What I did is I went over to "Entry Ramp Top Link Plate" file, and alter the "Sketch005" of a Body. Saved it, Went to "1-Entry RampAssembly" file, and I Updated the part. All good, and saved it. After I come to the "1-Frame-Assembly" file, press the "Update Imported Parts", it works fine,then I selected the Clone object named "1-Entry RampAssembly001". As soon as I pressed Del, this popped up. You can see it tried to delete the sketch I altered earlier. Do you think it is an Freecad issue?
[image]

kbwbe, 2018-12-30:
Hi polemidis,
as FC is trying to delete a sketch in your case, been selected some time in past, i think it is a FC core issue. Workflow of A2p does not track editing of sketches. I will have a look, whether i can prevent this after updateImportParts by actively deselecting everything.

Thank you for reporting !

kbwbe, 2018-12-30:
Hi polemidis,
i just enlarged the zoom factor of your screenshot to see more details. Why did you use a clone within the assemblyfile and not a simple A2p duplicate part ? I think the clone is in fact linked to something outside of assembly document and your problem is caused by these links. A2p assemblies containing **only** A2p objects work best.

polemidis, 2018-12-31:
> Hi polemidis,
> i just enlarged the zoom factor of your screenshot to see more details. Why did you use a clone within the assemblyfile and not a simple A2p duplicate part ? I think the clone is in [...]
I haven't think of the A2+/duplicate. I will try to see if this happens with that way, and I will try it with Freecad 0.17.

> Are you working with or without toponaming enabled ?

Disabled. I found some crazy stuff happening with that enabled. Btw, I use the A2+ daily, I can test experimental issues if you want to pin down bug or issues.

polemidis, 2019-01-17:
I am closing this issue, it looks like its more of a Freecad issue.

closed 2019-01-17 by polemidis