kbwbe/A2plus #164: Python bug on transparent view

donovaly, 2019-02-03:
I use 
OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.15780 (Git)
Build type: Release
Branch: master
Hash: f94e76c1116ec3b818961ff65131b3d86fa5b95e
Python version: 3.6.6
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

and the latest a2+ 0.2.9

When I click on the toolbar button "toggle transparency on an assembly" I get this python error:

Traceback (most recent call last):
  File "C:\Users\USti\AppData\Roaming\FreeCAD\Mod\A2plus\a2p_importpart.py", line 878, in Activated
    a2plib.setTransparency()
  File "C:\Users\USti\AppData\Roaming\FreeCAD\Mod\A2plus\a2plib.py", line 207, in setTransparency
    SAVED_TRANSPARENCY.append(obj.Name, obj.ViewObject.Transparency, obj.ViewObject.ShapeColor)
TypeError: append() takes exactly one argument (3 given)
Running the Python command 'a2p_ToggleTransparencyCommand' failed, try to resume

ManuelKrause, 2019-02-03:
Hi @donovaly,
Thank you for your report ! 
Maybe I'm in charge of this problem, as I changed that code section last.

May I ask you to upload a .fcstd assembly file in which this error occurrs? On here I can't reproduce it with my own assembly files, ATM.
Is there anything special about the assembly, that you could mention?

TIA

ManuelKrause, 2019-02-03:
Hi again, @donovaly 
just a shot into the blue:
Can you edit the file a2plib.py in your local Mod folder at the line ~207 in this way:

from:
                    SAVED_TRANSPARENCY.append(obj.Name, obj.ViewObject.Transparency, obj.ViewObject.ShapeColor)

to this three line version:
                    SAVED_TRANSPARENCY.append(
                        (obj.Name, obj.ViewObject.Transparency, obj.ViewObject.ShapeColor)
                    )

I don't know if this makes a difference, please try and report back.

TIA

donovaly, 2019-02-03:
> May I ask you to upload a .fcstd assembly file in which this error occurrs?

Simply create a block, cone or whatever using the part workbench.
Then change to A2+ WB and press the transparent view button while the block is selected

donovaly, 2019-02-03:
> Can you edit the file a2plib.py in your local Mod folder at the line ~207 in this way:

many thanks, this fixes the bug for me.

ManuelKrause, 2019-02-03:
Quite nice for you, but this exposes some new error for me on here:

Traceback (most recent call last):
  File "/home/manuel/.FreeCAD/Mod/A2plus/a2p_importpart.py", line 878, in Activated
    a2plib.setTransparency()
  File "/home/manuel/.FreeCAD/Mod/A2plus/a2plib.py", line 210, in setTransparency
    else:
AttributeError: 'Gui.ViewProviderDocumentObject' object has no attribute 'DiffuseColor'
Running the Python command 'a2p_ToggleTransparencyCommand' failed, try to resume

Oh, oh, another construction site...

Thanks for your feedback!

ManuelKrause, 2019-02-03:
@kbwbe 
I see a way to fix it. Give me one day...

ManuelKrause, 2019-02-05:
@donovaly 
Klaus, @kbwbe, merged my newest code into his devel-branch. Would it be possible for you to test this?

It should cover more corner cases around toggleTransparency than before. I've done my best to test with several assemblies, where it runs without errors.
It also works for your original use case (toggling transparency for a single part, not in an assembly) -- but there it still throws an error, although displaying works properly for me.
`Running the Python command 'a2p_ToggleTransparencyCommand' failed, try to resumeTraceback (most recent call last):
  File "/home/manuel/.FreeCAD/Mod/A2plus/a2p_importpart.py", line 898, in Activated
    a2plib.setTransparency()
  File "/home/manuel/.FreeCAD/Mod/A2plus/a2plib.py", line 224, in setTransparency
    obj.ViewObject.Transparency = 80
AttributeError: 'Gui.ViewProviderDocumentObject' object has no attribute 'Transparency'
Running the Python command 'a2p_ToggleTransparencyCommand' failed, try to resume`

If you have some time please test and provide your feedback from your machine.

I'm not sure on how to overcome this. @kbwbe ?

ManuelKrause, 2019-02-06:
Hi Klaus, @kbwbe 
sorry for more or less "hi-jacking" this thread...
After more testing, also with your commit of yesterday included, I also get the not selectable faces under certain circumstances. BUT: the aforementioned "AttributeError" looks like fixed !

Honestly speaking, this per-face-color and -transparency topic is still very confusing for me. I'm still in doubt that we handle this correctly in all places (unfortunately, already from the muxers onwards).
IMO, this thread originally regarding "transparent view" exposes the symptoms, but also has potential to find solution(s).

kbwbe, 2019-02-06:
Hi,
the primary python bug, leading to this issue report, is fixed with version 0.3.1

closed 2019-02-06 by kbwbe

donovaly, 2019-02-09:
Many thanks, now transparencies work fine.