kbwbe/A2plus #171: edit part fails for already opened parts

donovaly, 2019-02-09:
- open an FCStd file in FreeCAD
- create a new assembly
- add some pats, including the one that you have already opened in FC
- now highlight the part you have opened in the assembly and press the icon "edit"

result: nothing happens. In the console you see this:
```
Running the Python command 'a2p_editImportedPart' failed:
Traceback (most recent call last):
  File "C:\Users\USti\AppData\Roaming\FreeCAD\Mod\A2plus\a2p_importpart.py", line 617, in Activated
    FreeCAD.open(fileNameWithinProjectFile)
The project 'D:\usti\FreeCAD\CPP1\Pumpen-Body.FCStd' is already open!
```

Since assemblies consist of often dozens of parts, it is not a rare case that one part is already opened in FC. a2plus should therefore forward the info that the part to be edited is already opened. Either by output an info dialog or if possible, now info but switching simply to the part in FC (to its tab).

walterwissmann, 2019-02-09:
Yes, that would be helpfull :)

kbwbe, 2019-02-12:
It seems only to fail on Windows, on Linux it works.
I will try to make it working for windows too.

kbwbe, 2019-02-12:
This should be fixed with Version V0.3.5
Please update your A2plus WB.

closed 2019-02-12 by kbwbe