kbwbe/A2plus #173: missing documentation

donovaly, 2019-02-09:
I searched around and cannot find a proper documentation. I already wrote some docs for different programs and if you like, I could set up some Wiki pages here in GitHub or we ask the FC guys idf I can ass Wiki pages there.
I think a documentation would be helpful because e.g. for new users is is not clear why a2plus does not take into account that an already assembled part that is later on modified is not taken into account with the current shape but with the shape when it was assembled. Or it is not clear what is the most suitable way to replace a part in an assembly.

walterwissmann, 2019-02-09:
I work with 'MasterSketches' and LOT-technic.
The transparency cubes are placeholders to bind the part in the room.
To see all i switch off the mastercube inner the subassembly and switch on the part:
[image]
[image]

donovaly, 2019-02-10:
> I work with 'MasterSketches' and LOD-technic...

Interesting and I think these methods should be explained in the docs. If @kbwbe gives me a go I will set up at first the basic documentation and then write a special chapter for methods and concepts like yours.

walterwissmann, 2019-02-10:
the question is: on witch place should the wiki for a2+ come?

I think it should come to the FreeCAD-Wiki, not on github...

kbwbe, 2019-02-10:
@donovaly:
Here is the: GO!
Preferred place is the FreeCAD wiki, if we will get access.

ManuelKrause, 2019-02-11:
Hi @donovaly ,
you can find the preliminary documentation WIP at the bottom paragraphs of https://github.com/kbwbe/A2plus, it's in the README.md file. I haven't bothered with creating some WIKI style documentation so far, as I'm not experienced with it. 
I was quite impressed from the WIKI that can be found for the good old "assembly2" workbench here: https://github.com/hamish2014/FreeCAD_assembly2/wiki

But these words shouldn't interfere your creative process in any way.
All your help is definitely highly appreciated !
Best regards,
Manuel

donovaly, 2019-02-11:
I got the Go also from the FC devs. The documentation will be here:
https://www.freecadweb.org/wiki/External_workbenches

So I will start as soon as possible and of course take into account the info from
https://github.com/kbwbe/A2plus/blob/master/README.md

walterwissmann, 2019-02-12:
Hi @donovaly 
> > I work with 'MasterSketches' and LOD-technic...
> 
> Interesting and I think these methods should be explained in the docs. 

Here is an attempt to explain how to easily control complicated data structures with A2Plus.
https://forum.freecadweb.org/viewtopic.php?style=4&f=24&t=34101&p=286367#p286342
Please also scroll up and down...

It is not complete yet.
I have yet to represent the simple control options with A2Plus ...
I hope in the near future could be installed simple aids with A2Plus, which support the handling.

donovaly, 2019-02-17:
The documentation is almost ready: https://www.freecadweb.org/wiki/A2plus_Workbench#Part_Lists

@walterwissmann : your BIM work is very interesting and useful to know about but it is too complex for the Wiki page describing the basic A2plus stuff. Therefore please create a new Wiki page and make a tutorial out of it.
Here are nice tutorials where you can steal some Wiki formatting ideas:
https://www.freecadweb.org/wiki/Tutorials
and there is also a BIM tutorial:
https://www.freecadweb.org/wiki/Manual:BIM_modeling

You can get Wiki access here: https://forum.freecadweb.org/viewtopic.php?f=21&t=6830

closed 2019-02-17 by donovaly