kbwbe/A2plus #175: updated parts loose their color.

polemidis, 2019-02-10:
This is a new issue after the latest updates.

When you made some modification on an already imported part, save it , and update the parts, then it looses the color information. 
But if you delete that part, and re-import is everything looks fine. (other than reapplying the constraints). :) 
At the following image you can see that lever. I imported again and it carries the color information, but the existing one lost it and became grey.
[image]

walterwissmann, 2019-02-10:
That is an old issue, and i think we should give the guys a try ;)
They have done also a big work to the transparency issue

Have a look at the plug backside of my cnc controller:
[image]

This plug is a downloaded STEP-file and the frame should be blue. It comes through 3 subassemblies to the master assembly. The guys workes where fast and i think in 2 weeks we have this issue forgotten ;)

We can help the guy's, when we make a wiki ;)

polemidis, 2019-02-11:
walterwissman sorry man but I have hard time to understand what you are saying. I do not get how your comment helps in resolving this issue, and what you mean with the wiki.
This is where we post the issues so they can be fixed. An issue should not be taken personally as a criticism or whatever.

walterwissmann, 2019-02-11:
> This is where we post the issues so they can be fixed. An issue should not be taken personally as a criticism or whatever.
sorry, it should not be a critic to you or any else.
another sorry. my english is not the best...
br
Walter

ManuelKrause, 2019-02-11:
Hi @polemidis ,
thank you for reporting it right here. I don't think that @walterwissmann is completely wrong, as he adds the info, that it also occurs with multiple-level assemblies.
I'm aware of this issue and working on it to fix it. I need to be very careful in order to not mess up with the other aspects of color&transparency. It'll take some day or two until I can provide failure-free code, so I hope for your patience.
OTOH in the meantime, any other code fix proposal from you A2plus people is also highly appreciated!
Best regards,
Manuel

polemidis, 2019-02-11:
Manuel, no pressure at all man! I do not demand or expect anything! I am just reporting it. Thank you for your time and effort on this wonderful -and open source- workbench!! 
Waterwissmann no bad feelings, I just didn't understand what you said. My apologies if I sounded offending.
Thanks everyone for your contributions!

walterwissmann, 2019-02-12:
@polemidis,
We say in german 'Schwamm drüber' (swam over it) 
ok, shake hands, we have the same enthusiasm for FreeCAD and A2plus ;)

br
Walter

kbwbe, 2019-02-20:
Hi polemidis,
i started work on color topic. Sorry for waiting some time !

kbwbe, 2019-02-23:
Hi Polemidis,
color handling has been reworked at devel branch. Please test it.
Make sure, that the subassemby, which got grey during updating, has the property "updateColor=True" within the main assembly.

Please report new issue if the problem does not vanish.

closed 2019-02-23 by kbwbe

branilson, 2021-04-08:
Hello,

I am experiencing the same issue in a2plus over FreeCAD 0.19. The imported STEP files lost all face colors. While I was working on FreeCAD 0.18 certain faces loose their colors, but not everything as it is in the newer stable version of FreeCAD.

Do you have any suggestions on how to solve this?

kbwbe, 2021-04-10:
Perhaps it has to to with some preference settings of your FC step importer. Most settings which involve App::Link cause complete lost colors.