kbwbe/A2plus #178: inconsistent A2plus menu

donovaly, 2019-02-12:
The menu of A2plus is currently inconsistent:

[image]

1. The menu name should be "A2plus" not "A2p"

2. The menu entries should
- Begin consistently with capital letters
- Should be uniform in their name. E.g. not only "edit" but e.g. "Edit a part of the assembly"
- The naming should be consistent to the tooltip of the corresponding toolbar button

3. All menu entries should have a button. Currently the last entry "convert absolute paths..." has no icon. It should also appear in the A2plus toolbar.

kbwbe, 2019-02-12:
Is done with version v0.3.4

Last menu entry "convert absolute path..." got an icon, but was not put to the toolbar. It's use is seldom and sometimes dangerous. So i let the command only within the pull down menu.

Please update your A2plus WB.

closed 2019-02-12 by kbwbe

donovaly, 2019-02-12:
Many thanks! That was quick. the menu looks good.