kbwbe/A2plus #183: Constraint dialog appears sometimes outside of screen

donovaly, 2019-02-12:
When adding constraints the constraint dialog appears on random positions (which is not nice btw.). Sometimes the position is so that parts of the dialog are outside the screen and thus not visible. Here an example where the upper part is outside my desktop:
[image]

Sometimes the whole dialog is out of screen and then no dialog appears visibly.

If possible the constraint dialog should appear where the last opened constraint dialog appeared. The first appearing dialog of a session should appear in the center of the screen.

walterwissmann, 2019-02-12:
This was discussed for a while. 

It should pop up in the middle of the drawing window. Does it not?

kbwbe, 2019-02-12:
I will have a look where this problem comes from. The intended behaviour is:
1. opening on middle of FC Window
2. and following openings of the dialog near to last position, where it was placed by the user.

If it jumps around, this is definitely an ugly bug.

donovaly, 2019-02-12:
No it does not as you can see in my screenshot. This bug turns out to make a2plus hardly usable since now most of the time the dialog is completely out of screen.
Here is a good recipe to reproduce the bug:

1. create a constraint of your choice
1. select the constraint in the model tree and use the toolbar button "edit selected constraint"
1. do nothing else than accepting
1. repeat steps 2. and 3. and you can see that the dialog moves up every time and then it will be completely outside of the screen.

donovaly, 2019-02-12:
The bug is the same with FC "normal" and "Conda". After starting FC calling the dialog make it appear perfectly centered. The next call will place it a bit above the center and so on.

As workaround one can move the dialog back to the center before pressing accept. Then is slow move up will start from the position where you dragged the dialog.
Nevertheless this bug is super annoying at least for me while writing the docs and thus calling the dialog very often.

kbwbe, 2019-02-12:
A workaround, before i have found and fixed the bug is:
- open the constraint definition dialog (not the constraint property dialog)
- the position of the constraint definition dialog is fixed to the place where you put it.
- each constraint property dialog will open at the position of the constraint definition dialog.

So it does not move out of screen during your work.

donovaly, 2019-02-12:
> * each constraint property dialog will open at the position of the constraint definition dialog.

This doesn't work:
- open the constraint definition dialog
- place the dialog at the bottom
- close the dialog and call the constraint properties dialog
- call the constraint properties dialog subsequently and see how it moves to to the top and outside the screen.

kbwbe, 2019-02-12:
You have to leave the constraint definition dialog (constraint tool) open. Only then it works.
The opened constraint def dialog defines a master position for the constraint property dialog.

donovaly, 2019-02-12:
> You have to leave the constraint definition dialog (constraint tool) open.

This works. Thank you.

kbwbe, 2019-02-15:
Fixed on devel branch (version 0.3.10)

closed 2019-02-15 by kbwbe

donovaly, 2019-02-16:
Many thanks for the fix! This was important for Win users.