kbwbe/A2plus #184: constraint dialog has no default focus

donovaly, 2019-02-12:
- select a constraint in the model tree
- use the toolbar button to edit constraints
- you see that everything is fine and therefore you just press Return

result: nothing happens because the dialog has no button connected to have focus. The default focus should be on the Accept button. (pressing Return issues this button)

donovaly, 2019-02-12:
This also applies for the constraint definition dialog.

kbwbe, 2019-02-12:
Where to set best default focus at constraint properties dialog? Accept ?

At constraint definition dialog no default action is intended. Only help button could be possible.

donovaly, 2019-02-12:
> Where to set best default focus at constraint properties dialog? Accept ?

Yes, as I wrote in my initial comment ;-)

> At constraint definition dialog no default action is intended. Only help button could be possible.

Yes, the help button.

kbwbe, 2019-02-14:
I have set the ACCEPT button in constraint properties dialog as default button. This does not help as the dialog is not modal and often looses the focus. The non modal dialog was done to enable zooming/paning in 3D view during constraining, what is important.

Therefore i close this issue now. I do not see a practicable way to solve that at moment.

closed 2019-02-14 by kbwbe