kbwbe/A2plus #185: Contraint define dialog is always completely disabled

donovaly, 2019-02-12:
- open the constraint define dialog.

result: no matter what you have selected before opening the dialog, it is always completely disabled:

[image]

besides this, the dialog has also no default focus button. (Should be the help button)

kbwbe, 2019-02-12:
On my system (Linux Mint 18.3) buttons according to previous selection are activated.
Does this not work on Win ?

kbwbe, 2019-02-12:
This dialog stays on top of application until you close it. According to recent selection the suitable buttons get activated.

donovaly, 2019-02-12:
> according to previous selection are activated. Does this not work on Win ?

No, this doesn't work. I can select what I want but all buttons in the dialog remain greyed-out.

Also keeping the dialog open and select something does not help.

walterwissmann, 2019-02-13:
@donovaly 
i also have freecad  on win 7.

Take a click to the background and then click to the 2 things you will give the constrain. 
Then the previous selection should be activated

donovaly, 2019-02-13:
> Take a click to the background and then click to the 2 things you will give the constrain.

OK, then I get possible constraints in the dialog. And I only tried to select things in the model tree. -> a2lus needs documentation and I will document this now.

closed 2019-02-13 by donovaly