kbwbe/A2plus #186: highlight constraint has performance issues

donovaly, 2019-02-12:
- select a constraint in the model tree
- use the toolbar button to highlight the constraint

result: everything works fine, within a few milliseconds the assembly gets transparent and the constraint is highlighted.

- now you want to go back to the normal view and therefore left-click somewhere in the assembly

result: it takes here on my special and thus fast CAD PC 3 seconds until I get back the default view of the assembly. On slower machines it will take more seconds and might hinder the workflow.

kbwbe, 2019-02-12:
Perhaps this is an issue due to current work related to per face transparency. These issues are not completely solved at moment.

donovaly, 2019-02-12:
Ah, it is no performance thing but a Python error:

<unknown exception traceback><type 'exceptions.RuntimeError'>: maximum recursion depth exceeded while calling a Python object

ManuelKrause, 2019-02-13:
@donovaly, Hi!
It's quite likely that Klaus, @kbwbe, is right. At least my current investigations propose that with the current algorithms, for some objects at least, we don't catch the correct color tuple(s) properly, leading to later processing failures. In theory it's possible that then retrieving, throws your reported error.
Is there anything special you can tell about the object where it occurs for you ? (Or you upload it?)
The color&transparency thing is still quite tricky.
My goal is to unify the A2plus' color I/O to only depend on the color data given and not on the many A2plus' importable objects themselves.
It'll again take more time than I promised...

donovaly, 2019-02-13:
> Is there anything special you can tell about the object where it occurs for you

I sent it yesterday evening to @kbwbe by private mail. But as simple as it is, just create an assembly with 2 simple parts like an additive block. Then you can see the error in FC on Windows (no matter if with Python 2 or Python 3)

ManuelKrause, 2019-02-13:
@donovaly ,
thanks again! Sorry for your invested time...  I just saw it clearly by coincidental testing with my own parts/ assemblies.
You, posting the full error message, would have more easily taken out other errors...

`Traceback (most recent call last):
  File "/home/manuel/.FreeCAD/Mod/A2plus/a2p_importpart.py", line 904, in clearSelection
    if a2plib.isTransparencyEnabled() and not self.initialTransparencyState:
  File "/home/manuel/.FreeCAD/Mod/A2plus/a2plib.py", line 270, in isTransparencyEnabled
    return (len(SAVED_TRANSPARENCY) > 0)
<class 'RecursionError'>: maximum recursion depth exceeded in comparison´

It's in the color management. Sorry.

donovaly, 2019-02-13:
> You, posting the full error message, would have more easily taken out other errors...

I can only post what I get. I only get the single line I posted.

donovaly, 2019-02-13:
> I can only post what I get. I only get the single line I posted.

That is interesting: With the normal FC build I only get the single line I reported while only with the Conda build I get the full error message.
Thus I will have to use the conda build of FC for further work with A2plus so that I get more messages.

ManuelKrause, 2019-02-14:
Thank you for clarifying this difference. My openSUSE Tumbleweed system is more or less Conda like (except for the ?broken? icon and text sizes).

kbwbe, 2019-02-18:
Python recursion bug is fixed on devel branch. 2019-02-18
Please test again.

kbwbe, 2019-02-18:
> @donovaly ,
> thanks again! Sorry for your invested time... I just saw it clearly by coincidental testing with my own parts/ assemblies.
> You, posting the full error message, would have more easily [...]

Hi Manuel,
this error was caused by my code. The procedure restoreTransparency() was endless triggered by the selection observer.

I think the primary bug is fixed. I close this issue.

closed 2019-02-18 by kbwbe

donovaly, 2019-02-18:
Confirmed fixed, many thanks!