kbwbe/A2plus #189: axisPlaneParallel is in most cases not working

donovaly, 2019-02-12:
- select a linear edge and a plane
- create a axisPlaneParallel constraint

result: the edge will not be parallel to the plane. Both parts will stay as they are. I tried everything with different assemblies and constraint only works rarely for very simple assemblies. With no other constraint I have these problems thus i think there is probably a bug. I will send @kbwbe a test assembly by private mail where the constraints fails (does nothing) no matter what you try to use for the constraint.

kbwbe, 2019-02-14:
I just tested this constraint again and it is working on my side. I guess that none of your parts had a constraint path to a part with fixed position. If it is still not working for you, please PM me a file.
Thank you.

donovaly, 2019-02-14:
I already sent you a file via private mail 2 days ago. Have you got it?

kbwbe, 2019-02-14:
I do not find it. Please send again.

donovaly, 2019-02-14:
I just resent both to your kbwbe@ mail account. Please send me a reply mail that I know you got it.

kbwbe, 2019-02-14:
Yes, now i got some mails if files attached.
Thank you.

kbwbe, 2019-02-14:
I tested the file from your PM. As expected, no part was fixed. Close issue ?

donovaly, 2019-02-15:
Thank you! I'll close it.

closed 2019-02-15 by donovaly