kbwbe/A2plus #193: CenterOfMass direction rotates part

donovaly, 2019-02-13:
- create a CenterOfMass constraint
- in the appearing constraint properties dialog press a few times "Flip Dir."

result: the direction is correctly flipped but the part is also randomly rotated. This additional rotation is not necessary and only makes it more complicated to assembly further (at least in my tries).

kbwbe, 2019-02-14:
A direction flip can be done by every axis, which is vertical to the axis which is to be flipped. So there is an infinite number of possible rotation axis' for flipping. A2plus uses some random to flip an axis, in order to find a solution also in difficult cases. This cannot be changed for general cases. If rotation is not wanted, setting of more constraints is needed.

closed 2019-02-14 by kbwbe

donovaly, 2019-02-14:
OK.