kbwbe/A2plus #194: A2plus should use FC's file dialog

donovaly, 2019-02-13:
FC uses the native file dialog of the OS it runs on. A2plus should do this too if possible because some settings are OS-specific. Maybe A2plus can simply trigger the file dialog of FC to open/import parts.

donovaly, 2019-02-15:
Shouldn't changing line 335 of a2p_prefs.ui fix this issue?:
`<widget class="Gui::PrefFileChooser" name="fileChooser" native="true">`
So the change I did here:
https://github.com/kbwbe/A2plus/pull/209/commits/d33a82f886da4d23b5d1f3790dfbd834b1a06cd7

kbwbe, 2019-02-15:
> Shouldn't changing line 335 of a2p_prefs.ui fix this issue?:
> `<widget class="Gui::PrefFileChooser" name="fileChooser" native="true">`
> So the change I did here:
> [d33a82f](https://github.com/kbwbe/A2plus/commit/d33a82f886da4d23b5d1f3790dfbd834b1a06cd7)

No. This only affects the fileChooser of the preferences page and not the fileChooser for importing parts within impartPart.py

This is written in PySide Documentation:
`QFileDialog.DontUseNativeDialog | Don’t use the native file dialog. By default, the native file dialog is used unless you use a subclass of PySide.QtGui.QFileDialog that contains the Q_OBJECT() macro.`
According to documentation the native fileChooser is should be used by default. I am wondering if the fileChooser does this or not and why ?

donovaly, 2019-02-16:
I don't know PySide. Do you get under Linux the native file dialog or also the Qt file dialog? If you get the Qt dialog, what happens if you explicitly set the "native" flag for file chooser?

donovaly, 2019-02-17:
This is funny: when I use the Conda build of FC, I get Windows' native file dialog. On the "normal" FC build I get the Qt file dialog.

Since FC has to switch for all builds soon to Qt 5.x, there is in my opinion nothing to do here.

donovaly, 2019-02-17:
Just for the records, my Qt designer added the native flag to the class="Gui::PrefFileChooser" because I forgot to make this custom widget known to the designer. How this is done is described here: https://www.freecadweb.org/wiki/CompileOnWindows#Additional_stuff

So in future my changes to .ui files should work out of the box.

ManuelKrause, 2019-02-19:
With my Conda-like openSUSE Tumbleweed FC, I only wish, that 'most recently' used folders (/ history) would be recognized for the FC and A2-plus file-openers in the same way.

donovaly, 2019-02-19:
> I only wish...

I don't understand. The native file dialog should be used for all tasks. Because then you get what you want - if you changed the appearance for your system-wide file dialogs you also get it with FC and thus should also get it with A2plus. For the Conda build this is already the case, at least under Windows.

So you don't get the file dialog of KDE, Gnome or whatever you use?

kbwbe, 2019-02-23:
> Since FC has to switch for all builds soon to Qt 5.x, there is in my opinion nothing to do here.
Let's wait for FC switching to Qt 5.x.
For now, i close this issue.

closed 2019-02-23 by kbwbe

donovaly, 2019-02-24:
I fully agree.