kbwbe/A2plus #196: allow vertices in circularEdge constraint

donovaly, 2019-02-13:
The sphere constraint allows to select either:

* a spherical face and a spherical face
* a spherical face and a vertex
* two vertices

Therefore I see no reason why the circularEdge does not also allow:

* circular edge and circular edge
* circular edge and vertex
* two vertices

Putting the center point of a circular edge onto a vertex is useful.

kbwbe, 2019-02-13:
This is not ok. A circularEdgeConstraint does a position alignment and also an axis alignment. Last would not work with a vertex.

If such functionality is needed, it would be better to allow circular edges (center point) for the pointIdentity constraint.

donovaly, 2019-02-13:
> circularEdgeConstraint does a position alignment and also an axis alignment.

Ah, I see.

> If such functionality is needed, it would be better to allow circular edges (center point) for the pointIdentity constraint.

This is reasonable.
Note that I just reported things I wondered about. I can think of applications for such a constraint but this is no must-have.

kbwbe, 2019-02-14:
Please start new issue when functionality is needed.

closed 2019-02-14 by kbwbe