kbwbe/A2plus #197: conversion to a2plus part cannot be undone

donovaly, 2019-02-14:
- select an imported part and convert it to an a2plus part
- now you see that you selected the wrong part and thus want to undo the conversion

result: this is not possible.
If this is technically impossible for a FC addon, please close this bug.

kbwbe, 2019-02-15:
Fixed on devel branch. 2019-02-15

donovaly, 2019-02-16:
This is not completely fixed: 
 - convert a part to an a2plus part
- undo
- redo

result
`Traceback (most recent call last):
  File "C:\Users\USti\AppData\Roaming\FreeCAD\Mod\A2plus\a2p_observers.py", line 33, in slotRedoDocument
    a2p_importpart.a2p_repairTreeView()
  File "C:\Users\USti\AppData\Roaming\FreeCAD\Mod\A2plus\a2p_importpart.py", line 1100, in a2p_repairTreeView
    parent.Label = parent.Label # trigger an update...
<class 'AttributeError'>: 'NoneType' object has no attribute 'Label'`

kbwbe, 2019-02-16:
I only got this error when the converted part was already constrained. With the undo function these existing constraints loose their object reference.
I added a function to remove constraints related to non existing objects. For me it works now without error message.
Fixed within devel branch 2019-02-16

donovaly, 2019-02-16:
This is still not completely fixed or new issues have been introduced:

- open this simple assembly:
[testass3.zip](https://github.com/kbwbe/A2plus/files/2872005/testass3.zip)

* select the part "Pumpen-Kupplung_001"
* convert to A2plus part
* undo
* redo

result: info dialog stating "b_Pumpen_Kupplung_001_"

I see there 3 issues:

1. the first conversion to the a2plus part changes the name of the part (which makes sense) but the existing constraint is detached from the part and placed elsewhere to the model tree. Can't the constraint be kept?
1. the redo tries to restore the part with the original name, but since the conversion renamed the part, this fails
1. the info dialog outputs a part name that never existed.

kbwbe, 2019-02-16:
The test file does only contain A2p parts. What should be converted ?

donovaly, 2019-02-16:
> The test file does only contain A2p parts.

Not here on my PC. Both parts are simple imported parts.

> What should be converted ?

As I wrote  the part "Pumpen-Kupplung_001".