kbwbe/A2plus #198: toolbar button to delete constraints is always enabled

donovaly, 2019-02-14:
- select a constraint in the model tree
- use the toolbar button to delete constraints

result: you get the info that there is no constraint is connected.
This is sensible since the button deletes constraints from a part. Therefore the toolbar button should be deactivated when no part with an existing constraint is selected.

kbwbe, 2019-02-14:
fixed within devel branch. 2019-02-14

donovaly, 2019-02-14:
This is not fixed in A2plus 0.3.7:

- create a new assembly
- add 2 parts and not yet any constraint
- select a part in the model tree

result: the deletion button gets active. Click it and you get an error message. So the button should be disabled of no constraint is selected or if only a part is selected.

donovaly, 2019-02-14:
Ah, it is even worse: select a constraint and you see that the delete button is disabled.

kbwbe, 2019-02-15:
> Ah, it is even worse: select a constraint and you see that the delete button is disabled.

That's ok. The command is working on part level (to delete all constraints of it), not on a single constraint.

kbwbe, 2019-02-15:
Devel branch fixed again. 2019-02-15
Button now only gets active when selecting a part which is constrained.

closed 2019-02-15 by kbwbe