kbwbe/A2plus #199: deleting constraints make a sound

donovaly, 2019-02-14:
- use the toolbar button to delete a constraint

Result: you get a confirmation dialog. Unfortunately it appears with the Windows error sound. But there is no error. It is just the confirmation dialog for an action you purposely triggered.

kbwbe, 2019-02-14:
fixed within devel branch. 2019-02-14

donovaly, 2019-02-14:
This is not completely fixed in 0.37:

- create a constraint
- in the appearing constraint properties dialog press the button "Delete this constraint"

result: the confirmation dialog issues a sound despite no error occurred - it is just the confirmation of the action you purposely triggered.

kbwbe, 2019-02-14:
I tested again. I do not get sound on Linux Mint 18.3.
On your system, is it still the windows error sound or now another one ?

donovaly, 2019-02-14:
I use at the moment Windows 10 and I hear the Windows system sound for "Kritischer Abbruch".

donovaly, 2019-02-15:
Here is a PR to fix the issue: #210

kbwbe, 2019-02-15:
PR merged,
Thank you !

PR #210 (2019-02-15): fix issue #199 completely
- fix #199 completely
- improve and simplify wording
- whitespace
commit 0589c4f1 (2019-02-15): fix issue #199 completely

- fix #199 completely
- improve and simplify wording
- whitespace
closed 2019-02-15 by kbwbe