kbwbe/A2plus #203: HTML structure view fails for part names containing spaces

donovaly, 2019-02-14:
- create an assembly with 3 parts
- connect all parts so that every part has at least one constraint
- rename the first part to "partA", the second to "partB" and so on
- press the toolbar button to create the structure as HTML

result: everything is fine

- now rename "partA" to "part A" (note the space)
- press the toolbar button to create the structure as HTML

result: the HTML file doesn't show the structure anymore

Apparently the script at unpkg.com cannot handle spaces in names.

kbwbe, 2019-02-23:
Fixed at devel branch 2019-02-23
There was an unicode issue too. Part Labels containing german "Umlauts" did also not work for FC daily.
This is fixed too and it works now for FC0.18 daily and for Conda builds.

closed 2019-02-23 by kbwbe

donovaly, 2019-02-24:
Many thanks. Good to know that I can now use also umlauts in part names,