kbwbe/A2plus #205: A2plus fails if no part has the property fixed Position

donovaly, 2019-02-14:
I use the latest a2plus 0.3.7. I cannot solve this super simple assembly:

[2-part-testass.zip](https://github.com/kbwbe/A2plus/files/2866952/2-part-testass.zip)

there are only 2 parts, an axle and a wheel-like structure. Try to add a constraint between them of your choice.

result: the parts are not moved the solver only tells me:
`===== Start Solving System ====== 
TARGET   POS-ACCURACY :1e-05
REACHED  POS-ACCURACY :0.0
TARGET  SPIN-ACCURACY :1e-05
REACHED SPIN-ACCURACY :0.0
SA SPIN-ACCURACY      :0.0
===== System solved using partial + recursive unfixing =====`

The reason is that no part has gthe property "fixed Position" set to true. If the A2plus solver requires this, A2plus should
- either set one part of the constraint to fixed
- or does nothing but issue a dialog that at least one part must have fixed position.

As it is, average users like me are lost - it took me about an hour to realize what is going on.

kbwbe, 2019-02-14:
This is an old unsolved issue. I will try to find an algorithm which detects such constellations.
The first imported part is automatically set to fixed position. The problem appears e.g. when it is deleted.

easyw, 2019-02-15:
I would opt for raising a dialog. Doing something automatically would lead to not deaired behaviour...

kbwbe, 2019-02-15:
Introducing a flag "moved" to each rigid could be a solution.
If a constrained rigid does not have the flag set to value "True" after solving, this is an indicator for this problem. A dialog could be used in order to signal this.

kbwbe, 2019-02-16:
Now A2plus evaluates a new flag "moved" of each rigid.
After solving, parts with corresponding "rigid.moved == False" are highlighted and a dialog with a warning is shown.
Please update to version v0.3.11

donovaly, 2019-02-16:
Many thanks for this important improvement. However there are 2 issues:

- the information dialog is hidden
- although the constraint has no action it is created and the users has to delete it explicitly
Here a screencast:
[image]

My proposal:
- the info dialog appears on top
- the constraint is not created unless the user select nevertheless "Accept"

kbwbe, 2019-02-16:
Creation of constraint:
This is necessary, as the solver can only solve existing constraints. It is not really a preview. IMHO the normal case is, that the user only forgot to set an additional constraint to the fixed part. So he can keep the constraint if he wants this.

Hidden dialog.
This is a problem to solve. The constraint properties dialog is always pushed to front, so that the user can rotate and pan in the 3D view to inspect the result of the solving. This is conflicting with the new warning.

I hope i find a workaround.

donovaly, 2019-02-16:
And what about my second proposal? From my experience this would makes things a lot easier because assembling doesn't always follow purely logic paths  ;-) .

eduncan911, 2022-02-17:
Humm, I'm running into this today with two imported parts (step files) and have been unable to find a resolution.

I've even tried:

    >>> FreeCAD.getDocument('volocomosq').getObject('Solid').fixedPosition = True
    Traceback (most recent call last):
      File "<input>", line 1, in <module>
    AttributeError: 'Part.Feature' object has no attribute 'fixedPosition'

(my first python within freecad)