kbwbe/A2plus #206: Aa2plus proposes invalid constraints

donovaly, 2019-02-14:
- take this assembly:
[2-part-testass.zip](https://github.com/kbwbe/A2plus/files/2867042/2-part-testass.zip)
- look at the faces that are connected with the constraint
- select these 2 faces

result: in the constraint tools dialog and also in the toolbar A2plus offers you to set these constraints:
* angledPlanes -> this is an invalid constraint because 2 parallel planes cannot have an angle (OK the angle could be 0, but then the appearing constraint dialog must assure that the user cannot set another angel than 0.
* planesParallel -> this constraint does already exist and a duplicated constraint should not be proposed if possible 

I guess this is not easy to fix. So maybe this should be added to the feature wishes list.

kbwbe, 2019-02-14:
This is indeed not easy to fix. Conflicting constraints are only detected during solving, as the complete constraint system gets not solvable. In complex assemblies, it is even nearly impossible to detect which constraint caused this.  So ATM it needs some care of the user to apply the constraints and check the result at once.

A2plus is very robust against redundant constraints, in opposite to solve space.

kbwbe, 2019-04-29:
I close this issue.
Calculations during setting of a constraint assuring that it is valid/solvable are much to complex and cannot be done here.

closed 2019-04-29 by kbwbe