kbwbe/A2plus #211: 'German Umlaut'

walterwissmann, 2019-02-15:
@donovaly 
please add at Troubleshooting that filenames with 'German Umlaut' can make problems with false installed ascii tables with python...

They should not take { Ä, ä, Ö, ö, Ü, ü, ß ) in the Filenames.
They should take (Ae, ae, Oe, Ue, ue, ss )

I have a huge over 2 days searching backside me to find this issue...

Regards 
Walter

donovaly, 2019-02-15:
I can add this. The question is if this issue here will be fixed before the Wiki page will be announced.

The Wiki page is not yet ready, the preferences and the parts lists are not yet described.

kbwbe, 2019-02-15:
Encoding issues regularly cause problems. I tried to avoid this but it seems to be very difficult, especially if working with different systems.
So an entry within troubleshooting section of documentation makes sense.

walterwissmann, 2019-02-15:
Ok, I see that the information is known to you;) You can close this issue if you think so.

Regards,
Walter

donovaly, 2019-02-16:
I added now a note in https://www.freecadweb.org/wiki/A2plus_Workbench#Troubleshooting

closed 2019-02-16 by donovaly