kbwbe/A2plus #224: enhancement: clicking on constraint opens properties dialog

donovaly, 2019-02-17:
It would be a useful feature if one could double-click on a constraint in the model tree and the constraint properties is opened.

kbwbe, 2019-02-17:
Done within devel branch. 2019-02-17

donovaly, 2019-02-17:
There is a bug with this feature:

- open an assembly
- switch to another WB (not A2plus)
- double-click on a constraint

result:
`Traceback (most recent call last):
  File "C:\Users\USti\AppData\Roaming\FreeCAD\Mod\A2plus\a2p_viewProviderProxies.py", line 174, in doubleClicked
    FreeCADGui.runCommand("a2p_EditConstraintCommand")
<class 'Base.FreeCADError'>: No such command 'a2p_EditConstraintCommand'
ConstraintMirrorViewProviderProxy doubleClicked`

So when the users double-clicks we must at first load the A2plus WB.  (This is also done for other WBs, for example when making a body or a FEM analysis active).

donovaly, 2019-02-17:
There is another minor issue:
whenever I double-click on a constraint I see now in the report panel of FC a line like this:
"ConstraintViewProviderProxy doubleClicked"
Is this really necessary? i mean nothing special happened, no warning, no error, A2plus did just what the users wanted.

kbwbe, 2019-02-17:
> There is another minor issue:
> whenever I double-click on a constraint I see now in the report panel of FC a line like this:
> "ConstraintViewProviderProxy doubleClicked"
> Is this really necessary? i mean nothing special happened, no warning, no error, A2plus did just what the users wanted.
Sorry, i forgot to remove some debug prints.

kbwbe, 2019-02-17:
The workbench is now automatically activated during evaluating the double click.
(Devel branch)
Please test again

donovaly, 2019-02-17:
I tested and the bug is now fixed. many thanks!

closed 2019-02-17 by donovaly