kbwbe/A2plus #229: Fresh installation has uncomitted Git changes

donovaly, 2019-02-17:
on Windows
- uninstall the A2plus WB
- restart FC and install it again

A2Plus was installed in the user folder, in my case under
C:\Users\USti\AppData\Roaming\FreeCAD\Mod\A2plus

Git tells me that this folder has uncomitted changes. I viewed the DIFF and see this:
[image]

These are the 2 SVG files I changed today. I have no clue why their commits are not taken into account because I made a normal PR when I changed the files.

kbwbe, 2019-02-23:
This seems to be a problem on your local site.
A git clone https://github.com/kbwbe/A2plus.git directly to my mod folder does not show any uncommitted changes.
Installing via the addon manager does not create a valid git repo on my side. This seems only to copy the files to the mod folder.
In either case, i do not find uncommitted files on a fresh installation.

donovaly, 2019-02-24:
> This seems to be a problem on your local site.

No. I now used a laptop that has never seen FC yet. I used the latest FC 0.18 build and there installed A2plus. As result I get again a changed tree:
[image]

Note there the date of the change - it was done directly while installing A2plus. So the installation routine in FC triggers to fetch A2plus from GitHub but the 2 SVG icons are marked as changed. Therefore I receive them as changed too.
This is at least my explanation and that makes me sure that there is an issue on the A2plus side.

However, it can of course be neglected from the user's perspective. I only wanted to make sure I did not break anything with my commit that changed the SVG files. So please close the bug again of you say we can ignore it.

kbwbe, 2019-02-24:
I checked again. 
1.) Method: git clone:
[image]
So git clone works for me.

2.) addon manager:
[image]
All svg icons have the same date.

So everything works as expected on my side (Linux Mint Sylvia 18.3). It does not seem that something is broken with Github. I close this issue again.

easyw, 2019-02-24:
@kbwbe 
I forked your repo after deleting mine and I also get uncommitted changes
[image]

kbwbe, 2019-02-24:
Thank you, @easyw 
I am wondering how this could happen ?

easyw, 2019-02-24:
Git mystery :smile:

donovaly, 2019-02-24:
Sorry to reopen but this commit today broke the SVG files I fixed:
https://github.com/kbwbe/A2plus/commit/874ead1ea8042e383a45b01958e41fac1ef86c6f

You restored the old files but I had to change them because they were broken in respect that browsers like Firefox couldn't display them, see e.g.
https://www.freecadweb.org/wiki/File:A2p_PartsInfo.svg

Moreover you restored their incorrect Sodipodi names.

Please revert this commit or fix the SVGs they way you like it, but don't leave them broken.

kbwbe, 2019-02-25:
@donovaly,
if possible, please PM me the correct ones. I am not sure which ones are OK.

donovaly, 2019-02-25:
Thank you, the icons from the Wiki page are the corrected ones.
Lets see if this time the commit went through in the form that git doesn't tell me there are uncommitted changes. I would therefore like to keep the bug open until I could test or until the next release.

donovaly, 2019-02-26:
> I would therefore like to keep the bug open until I could test or until the next release.

I checked and this issue is now fixed in version 0.4.0

closed 2019-02-26 by donovaly