kbwbe/A2plus #230: preferences are broken

donovaly, 2019-02-18:
- use a fresh installation of A2plus (uninstall existing A2plus before)
- restart FC
- change to e.g. the Part WB (assure that A2plus was never active after you started FC)
- use the menu Edit->Preferences

Watch the report panel output and note that you can do changes without any problems.

- close the preferences dialog
- activate the A2plus WB
- open Preferences again

result: in the report view panel of FC you see
"Cannot restore!"

- now uncheck the option to show constraints in the toolbar and press apply

result: the toolbar buttons are still shown and in the report panel you get:
"Cannot save!"

donovaly, 2019-02-18:
In the file user.cfg I see that this line changed correctly:

`<FCBool Name="showConstraintsOnToolbar" Value="1"/>`

So something was actually saved but it is not taken into account by A2plus. Only when i restart FC i can see my changes 8the toolbar buttons are not shown)

No other WBs are affected thus I am quite sure that it is a A2plus problem.

donovaly, 2019-02-18:
The bug is also in A2plus 0.3.10.

kbwbe, 2019-02-18:
It seems to be the filechooser. It is not possible to set a path... !! ??

donovaly, 2019-02-18:
> It seems to be the filechooser. It is not possible to set a path... !! ??

So did I break it?
Currently I cannot set a path because of this bug.

kbwbe, 2019-02-18:
No. I think it is something more general.

kbwbe, 2019-02-19:
This is fixed with version 0.3.16.
Thank you for reporting.

donovaly, 2019-02-19:
So one problem was that one needs to tell the Qt creator about the custom FileChooser widget:
https://www.freecadweb.org/wiki/CompileOnUnix#Qt_designer_plugin
before one can edit the .ui file.

The bug is however not completely fixed:
- uncheck in the preferences the toolbar buttons for the different constraints and press Apply

result: nothing happens, you still see all toolbar buttons. Only after restarting FC you will see the buttons no longer.

luzpaz, 2020-03-22:
does this issue still persist ?

luzpaz, 2021-02-28:
@kbwbe ping

kbwbe, 2021-02-28:
The issue still exists.