kbwbe/A2plus #232: improve handling of angledPlanes constraint

donovaly, 2019-02-19:
The vast majority of parts are constructed according to one of the 3 base planes (xy, xz and yz). Therefore when importing parts to A2plus many faces have to each other an angle of 0, 90, 180 or 270°.

The problem is that setting an angledPlanes constraint fails when the input angle between 2 planes is 0 or 180°. Take for example this simply assembly and see that the solver cannot solve the constraint:
[testass3.zip](https://github.com/kbwbe/A2plus/files/2879750/testass3.zip)

To fix this, the user can rotate one part by few degrees and then the solver finds a solution. But this information requires a lot ob background knowledge how solvers work. Therefore A2plus should do this trick automatically in the background:
- the user wants an angledPlanes constraint
- A2plus knows the current angle of the selected planes. If it is 0 or 180°, it will
 - either in the background or visibly rotate the second part of the constraint by 5° in one of the directions of the plane (if it is e.g. a plane in x-y, the either rotate around the x- or the y-axis.
 -or maybe it is even enough just not to pass a 0 or 180 to the solver as input angle but instead 5 or 185, respectively.
- If the solver succeeds with this, the parts are moved accordingly, if not the parts stay like they are but A2plus outputs then an information dialog telling that rotating the parts a bit manually could solve the problem.

kbwbe, 2019-02-24:
This should be done at devel branch. 2019-02-24
Please test again.

donovaly, 2019-02-26:
> Please test again.

Great! It works fine here with my test cases. This is a big improvement for the usability.

So in respect to all your recent improvements and fixes, maybe one of the next released can  be "0.4" or from my point of view even a "0.9" release. I mean A2plus works well now for real-life assemblies. 
From my point of view it would be nice to set up a defined movement, for example letting an offset or angle change between 2 defined values to visualize a movement later in reality.

donovaly, 2019-02-26:
Fixed in in version 0.4.0

closed 2019-02-26 by donovaly