kbwbe/A2plus #237: enhancement: allowing LCS import and contraints

easyw, 2019-02-24:
Hi @kbwbe I'm adding a note here as we have talked in PM.
It would be nice to allow A2p during import a part to include also LCS(s) if they are inside the part to be imported.
The LCS imported should be with placement attached to the related A2p object.
A2p could then use LCS to create a new constraint which will overtake the topology issues even on a standard FC main code...

walterwissmann, 2019-02-24:
what is 'LCS' ?

BR
Walter

easyw, 2019-02-24:
@walterwissmann 
LCS -> Local Coordinate System
you can search the FC forum for a deeper insight
https://www.google.com/search?q=freecad+forum+lcs

easyw, 2019-02-24:
@walterwissmann 
LCS -> Local Coordinate System
you can search at FC forum for a deeper insight
https://www.google.com/search?q=freecad+forum+lcs

walterwissmann, 2019-02-26:
Thank you, easyw :)

BR from
roerich_64 ;)

kbwbe, 2019-03-02:
@easyw , please check PM.

meprakarsa, 2022-02-18:
I have similar wish here...
Since the first time I used A2plus (last year :) )