kbwbe/A2plus #253: A2plus "move" doesn't use current windows' center focus

ManuelKrause, 2019-03-20:


kbwbe, 2019-03-20:
Hi Manuel,
if using the move button the part connects to the mouse pointer. When you are moving back from hitting the button the part will first appear in top left corner of 3D view (and not there where you have placed it before). This is normal.

If you do not want to have this effect, please use the FC built in part mover by double clicking the part in the tree view. Then it stays in place until you move it explicitly.

As therefore a solution exists within FC, i will not work on this topic.

closed 2019-03-20 by kbwbe

ManuelKrause, 2019-03-21:
Hi Klaus,
thank you for your explanation. This helps a lot !  I just wasn't aware of this solution.
Then you're absolutely right to not bother with this topic.