kbwbe/A2plus #271: A2plus fileChooser filter is fixed to mixed case, cannot select 0.17 files with lowercase suffixes

glabifrons, 2019-04-29:
Issue:
When attempting to import a part into an assembly ("Add a part from an external file"), the fileChooser dialog has only one file type option "Supported Formats" (without defining what those are).
There is no alternate "All Files"/"*" catch-all/fallback option available.
None of my files are shown (with 185 *.fcstd in the directory).
If I create a link or rename a file to have an FCStd suffix instead, A2plus then can see and load that file.
It is also case-sensitive for the stp & step suffixes. Some files, when copied from other platforms, may have uppercase suffixes. These cannot be seen by A2plus's fileChooser, but files with all-lowercase stp & step suffixes can.

Expected behavior:
1. The fileChooser should be case-insensitive to match FreeCAD itself. In FreeCAD, I can open files with both all lowercase and mixed-case filename suffixes, and it previously (prior to 0.18) defaulted to saving with an all lowercase suffix.
2. The formats "Supported Formats" includes should be shown (like in the fileChooser FreeCAD uses).
3. An alternate file type selection "All Files"/"*" should be available.

I tried both the version from the freecad-maintainers PPA and the latest AppImage from freecadweb.org.

PPA version:
OS: Ubuntu 18.04.2 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.1.
Build type: Release
Python version: 2.7.15rc1
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

AppImage version:
OS: Ubuntu 18.04.2 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16110 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.18.1)
Hash: f7dccfaa909e5b9da26bf50c4a22ccca9bb10c40
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)

The closest I could find is: https://github.com/kbwbe/A2plus/issues/194

Please let me know if you need any more info.

kbwbe, 2019-04-29:
Thank you for reporting. I will check and fix this.

kbwbe, 2019-04-29:
The problem is a misbehavior of the builtin file dialog of Ubuntu 18.04 LTS. I could reproduce the error on Linux MINT 19.1, which is based on Ubuntu 18.04 LTS.
A2plus now forbids the use of operating systems native file dialogs and uses Qt's build in ones.

Please update A2plus to version V0.4.18 and test again. On my system, it is working now.

glabifrons, 2019-04-30:
Wow! That was fast!
Yes, that works nicely, all issues addressed.
Thank you very much!

closed 2019-04-30 by glabifrons