kbwbe/A2plus #287: rotates (or maybe moves) only to certain degree

stingray427, 2019-06-09:
Hello A2plus Team,

so the symptoms are:
- for example values from 90 to 59 work for constraint "angledPlanes_001__base_002" between arm and fixed base-part.
- at value of 58 it doesn't work and I get a warning see the picture. "Constraints inconsistent. Cannot solve System.  Please delete your last created constraint !"
- also value 59 works, but it takes really long time to compute. It takes about 5 seconds.
Okay I have an extremely old & slow CPU 3570k @4.0 Ghz, but solving at 82 degrees takes probably less than 0.5 seconds. So there is difference 10-folds.

I thought converting every part into "full function A2plus part" would work, but as you see in this case it doesn't help.

My suspicion is,  as soon rod's (yellow) center and tube's (pink) center cross solver stops to work :(

[parts.zip](https://github.com/kbwbe/A2plus/files/3269667/parts.zip)
[image]

kbwbe, 2019-06-15:
Hi,
sorry for the my late answer, i am just back from a few days on holiday.
The problem you encountered is, that going close to 58 degrees the iteration steps of the solver get instable. One indicator is the huge calculation time when using the 59 degree value.

The reason probably is the constraining of the rod. It's bearing sleeve is far away from the rotation axis of the spring mount base.

Please find attached a version, where i changed the constraining of the rod. Please inspect axisCoincident_003... and axisConincident_004...

With this constraint sceme it works with 58 and beyond degrees. I tested it up to 54 degrees.
[parts.zip](https://github.com/kbwbe/A2plus/files/3293003/parts.zip)

[image]

stingray427, 2019-06-16:
No worry, I wasn't expecting at all that you will provide me a workaround. Even though I found another workaround(not very good one), I am still very impressed by your kindness. It is still issue tracker not a forum after all. Thank you! :-)

Suppose it is a bug and not a limitation of mathematics, I provide my another workaround for this problem in case someone wants to study this case more closely.

This is what I did: i have changed lengths of the rods and bearing sleeve.

- in file "assembly_test_a2plus.FCStd" there is long bearing sleeve, which let solver break already at 66 degree.

- in file "assembly_test_a2plus_short_bearing.FCStd" I use shorter sleeve bearing. Works also with 54 or even 45 degrees.
[range_test.zip](https://github.com/kbwbe/A2plus/files/3294565/range_test.zip)

kbwbe, 2019-06-17:
This is an unexpected effect and i can comfirm the behavior. Fixing this will for sure need some time as such things are usually difficult to explore. I let this issue open as reminder.