kbwbe/A2plus #289: Missing objects to import from FCStd

sharpener13, 2019-07-14:
Due to multiple FC complications, I tend to use only few methods to create objects in FC:
- sketch (Sketcher)
- extrude (Part)
- boolean (Part)
- clone (Draft)

Unfortunately this leads to issues with A2+. Probably the Draft's clone causes the objects used as a clone source are not shown when trying to import single objects from FCStd (select dialog shown, found object shown, missing some of the objects from FCStd).

There are two objects in the example attached: `HeadBasic` and `HeadAdvanced`. You might want to try to import `HeadBasic`, but it is not shown in the selection dialog. If you delete the `HeadAdvanced` from the FCStd file and save it `HeadBasic` shows up.

Using:
- latest FC 0.19-pre (17352)
- latest A2+ 0.4.25

[image]
[a2plus-example.zip](https://github.com/kbwbe/A2plus/files/3390242/a2plus-example.zip)

sharpener13, 2019-07-20:
Any thoughts or workarounds?

kbwbe, 2019-07-21:
I can reproduce this issue with your example file. Doing a simple clone is working. But if a feature is added to the clone it's original disappears. I will check the code for this.

kbwbe, 2019-07-21:
Please update your A2plus WB to version V0.4.26. This should fix the problem. Please test again.

sharpener13, 2019-07-21:
Cool. It works as expected. Thank you!

closed 2019-07-21 by kbwbe