kbwbe/A2plus #292: Ordered Workbench list

leoheck, 2019-07-28:
Hello!

First of all, this plugin is awesome, thank you for the efforts to make it really good.

I saw that many plugins, when added to Freecad are inserted on the Workbench list in an ordered manner. This one, however, is being added to the end of the list. Is it possible to update it to do the same? It is faster to select the workbench when the list is ordered.

Examples of plugins that are being inserter in an ordered way
- Fasteners
- Part-o-magic

leoheck, 2019-07-28:
I think this should be done like this: https://github.com/shaise/FreeCAD_FastenersWB/blob/b3cd9b48af8ba448451f5a3f43b8096c2ab6e283/InitGui.py#L64

This is the Part-o-magic setup.
https://github.com/DeepSOIC/Part-o-magic/blob/master/InitGui.py

kbwbe, 2019-07-28:
Hi, thank you for your hint. The marked line within Fasteners WB is already present within A2p. But it does not seem to influence the order of the WB list. So i do not really know where to change the behavior.

As a tip for switching between WB's: I use the TabBar addon of @Triplus. You can activate your favorite WB's and switch between with only one click.

[image]

Turro75, 2019-07-28:
Dear Leoheck,

if You right click over the toolbar You can reach the menu editor, in the third tap You can manually set the order of the workbenches and which can be listed or not.
A2plus will respect that order.

Best Regards

P.S.

Ciao Klaus!

leoheck, 2019-08-01:
@Turro75 sure, I know this, but it would be really nice to have this in order by default, isn't it? I can try to help to improve this if you accept the effort.

leoheck, 2019-08-01:
@kbwbe cool I will try to debug it, lets see.

triplus, 2019-08-12:
Once you open Tools -> Customize ... and click on a Workbenches tab a list of workbenches is saved (Tools -> Edit parameters ... -> BaseApp -> Workbenches). Installing new modules will add them to the end. This is current standard behavior and likely it doesn't make much sense for individual modules to try to override it.

P.S. If you want to get default sorting back. Delete the Enabled item through parameter editor and restart FreeCAD.

easyw, 2019-08-12:
The normal behavior is to add the new WB in alphabetical order.
With this [PR](https://github.com/easyw/kicadStepUpMod/commit/b34810860e1bfdf74c860823164a834370540a8e) I handled to do it also for StepUp WB.

leoheck, 2019-08-13:
Can confirm, @easyw PR is working.

closed 2019-08-21 by kbwbe

leoheck, 2019-08-21:
Awesome, thanks dude, it is working.

PR #296 (2019-08-21): Devel
fixes #292 Ordered Workbench list