kbwbe/A2plus #293: Option to extend holes across matching faces

steveinjava, 2019-08-09:
Any holes in the mating faces are quite likely for anchoring. If any holes in those faces could be offered to be continued- that would really help me for sure and probably many others? What do others think?
I am designing a CNC router and I have the base with a linear bearing which I need to attach a 4040 extrusion to. The bearing has threaded anchor holes. It would be great if the 4040 could be 'drilled' to match without doing the maths...
[image]

silopolis, 2019-08-10:
I think the best solution to this problem is having subtractive objects
that can be added to components and create the negative features needed by
other features/parts/components/assemblies.
Imagine a bolt component combining the part together with the tapped hole.
When you place the component, intersecting parts are "drilled".
Ideally, some kind of scope would be selectable for the subtractive feature
to define whether it should apply to parts in the same component or
assembly or all parts it crosses.
Have a look at CadQuery and CQParts for an example of implementation.

>

steveinjava, 2019-08-10:
Thanks. I was thinking I could introduce some fixings to the original
object "through" the new attachment. Even just doing that would be okish- I
could create a techdraw from it with dimensions for the obvious hole
needed. I will have a play around at using booleans but I don't actually
want the attachments holes to be threaded- just clearance holes. Anyway, I
will see what can be done with existing facilities. Thanks again, Steve

closed 2019-10-09 by kbwbe