kbwbe/A2plus #298: Failed to import part (object has no attribute 'DiffuseColor')

justacec, 2019-10-09:
I have a quick part that I put together to represent the VMA208 Accelerometer board.  When I tried to import it into an assembly, nothing shows up in the screen and the following error message is displayed in the report view:

```
Running the Python command 'a2p_ImportPart' failed:
Traceback (most recent call last):
  File "/Users/justaceclutter/Library/Preferences/FreeCAD/Mod/A2plus/a2p_importpart.py", line 567, in Activated
    importedObject = importPartFromFile(doc, filename)
  File "/Users/justaceclutter/Library/Preferences/FreeCAD/Mod/A2plus/a2p_importpart.py", line 308, in importPartFromFile
    topoMapper.createTopoNames()
  File "/Users/justaceclutter/Library/Preferences/FreeCAD/Mod/A2plus/a2p_topomapper.py", line 516, in createTopoNames
    needDiffuseExtension = ( len(ob.ViewObject.DiffuseColor) < len(ob.Shape.Faces) )

'Gui.ViewProviderDocumentObject' object has no attribute 'DiffuseColor'
```

Anything thing that I find strange is that the mux info is blank.  Now I admit that I really have no idea what the mux info property is holding, but it being blank seems bad...

I have attached the resulting A2Plus file and the part that I was attempting to import.  Any feedbag would be great.  I know there were some other errors related to DiffuseColor but they seem to be a bit different.  

To replicate this simply create a blank document, switch to the A2Plus WB and then add the attached VMA208 file though the A2Plus WB.

My setup is the following:
OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18403 (Git)
Build type: Release
Branch: master
Hash: 0717b4fc23ef1db70964c3977d25e2fe46a739d1
Python version: 3.7.3
Qt version: 5.12.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/United States (en_US)
A2Plus: 0.4.26b

Interestingly enough, you cannot upload the regular Freecad files here.  Because of this, I zipped up both files into a single archive.  Let me know if this did not work.

[failed_assembly.zip](https://github.com/kbwbe/A2plus/files/3706991/failed_assembly.zip)

kbwbe, 2019-10-09:
Hi, thanks for reporting.
I just did a quick test using FC0.18.1.
With that FC version the import works as expected. I guess, it has to do with changes related to FC0.19 (link merge).
I will try to do some more tests this evening with FC0.19

luzpaz, 2019-10-09:
JFYI @kbwbe v0.18.3 is the stable not 0.18.1

kbwbe, 2019-10-09:
thanks @luzpaz. I know. At work, i am on 0.18.1, at home i am on 0.18.3 and recent 0.19 master branch.

kbwbe, 2019-10-09:
FC0.18.3 works.
FC0.19.18405 does not work and i can reproduce the described error.

There seems to be a change within the objects hierarchy, depending on FC version. This causes struggling of the import part algorithm when using FC0.19. 

I have to investigate what is happening there.

kbwbe, 2019-10-09:
Please update your A2Plus version to V0.4.27 and test again. It should work now.
The problem was a changed data structure of the part container within FC0.19 after link merge.
Please give a feed back.

maximbendiks, 2019-10-10:
With V0.4.27 I still have the same error when importing a step file. Unfortunately I can't share the file but I will try to reproduce it with some other file.

OS: Manjaro Linux (dwm/dwm)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18504 (Git)
Build type: Release
Branch: master
Hash: f66023a646db4b2502bb3637897443b3525ca3c7
Python version: 3.7.4
Qt version: 5.13.0
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

maximbendiks, 2019-10-10:
I created a random block in part design, exported it to step and tried importing it with a2plus. It does not get imported at all (but also no DiffuseColor error) and I get following message: 
```
<Import>` AppImportGuiPy.cpp(520): file read time: 0.0115175s
<Import> AppImportGuiPy.cpp(521): import time: 0.007585s
<Import> AppImportGuiPy.cpp(522): total time: 0.0191025s
imported Object is empty/none
```
I also tried putting the body container of the block into a part container, exporting to step and import with a2plus which failed same as above.

When I drag the body of the block into my a2plus assembly and try to convert it, it works but I get following message which says it failed:
```
Running the Python command 'a2p_ConvertPart' failed:
Traceback (most recent call last):
  File "/home/maxim/.FreeCAD/Mod/A2plus/a2p_convertPart.py", line 164, in Activated
    convertToImportedPart(doc, selection[0])
  File "/home/maxim/.FreeCAD/Mod/A2plus/a2p_convertPart.py", line 72, in convertToImportedPart
    setattr(newObj.ViewObject, p, getattr( obj.ViewObject, p))

'Gui.ViewProviderDocumentObject' object has no attribute 'DisplayModeBody'
```

[a2plus_test1.step.zip](https://github.com/kbwbe/A2plus/files/3713942/a2plus_test1.step.zip)

kbwbe, 2019-10-11:
I will check this...

kbwbe, 2019-10-12:
I can not exactly reproduce your errors which you described above. But step import was definitely not ok. My previous fix was not sufficient. I did some changes and after that step import worked for me again.
Please update your A2plus WB again and test the new version V0.4.27a

My system is:
OS: Linux Mint 19.2 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18405 (Git)
Build type: Unknown
Branch: master
Hash: c6d1684e2a84f9487c232fa38ea11b8b84654636
Python version: 3.6.8
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

maximbendiks, 2019-10-14:
Great! It works now. I will do further testing. Thank you so much.
The only thing I discovered so far is that the selection of faces and edges in the assembly file has a quirk. The first selection gets highlighted but the selection after that with Ctrl does not, though they are selected properly and I can create constraints. When I open a simple file with a body container the highlighting works.

OS: Manjaro Linux (dwm/dwm)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18513 (Git)
Build type: Release
Branch: master
Hash: fd352c14f13ecdef7b8d4005c987d2f449c69430
Python version: 3.7.4
Qt version: 5.13.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

scholi, 2021-03-10:
Hi and thank you for the nice work. A2plus is really great. Unfortunately I have the exact same issue with latest a2plus. Can someone help please?

OS: Windows 10 (10.0)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.23970 (Git)
Build type: Release
Branch: master
Hash: a30cf5b4df22cf43f58211210b70183a7d6c17aa
Python version: 3.6.8
Qt version: 5.12.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: French/France (fr_FR)

A2plus git hash: 2e38ed41c77219fc44aa537e5bf6954abd4f35f3

kbwbe, 2021-03-10:
Do you have a file for me ? Please upload one.

scholi, 2021-03-15:
I could find that the issue comes with parts having this icon: 
[image]
I think those are links that have the property "Shape Material" which itself has the problematic "Diffuse Color". My file is a bit big I'll prepare a simplified buggy version

scholi, 2021-03-15:
Here is an example
[bug.zip](https://github.com/kbwbe/A2plus/files/6139938/bug.zip)

kbwbe, 2021-03-15:
@scholi , thanks for the file.
I cannot import it to A2p, but the error which is shown does not deal with "Diffuse Color".  A2p is struggling with the object tree structure at all. I get this:
```
File "/home/klaus/.FreeCAD/Mod/A2plus/a2p_topomapper.py", line 274, in populateShapeDict
    pl = ob.getGlobalPlacement().multiply(ob.Placement.inverse())

'App.DocumentObject' object has no attribute 'getGlobalPlacement'
```
Maybe this is caused by the "Link Group" objects, which you mentioned above. I will try to debug this, when there is time.
Perhaps at weekend.

Why is it not possible to model your connector (J2_M55-7008042_ASM_60215747 )in a separate file and assemble it there ? Then the complete connector could be used as a sub assembly in the main assembly of the board. Where is the advantage of using link groups ?

scholi, 2021-03-16:
I'm actually not sure why this happens. I personnaly never use links. It is the KiCadStepUp module that creates the part for me from a PCB design CAD (KiCAD). The solution for me at the moment is to use KiCadStepUp on FreeCAD 0.18 export it to STEP file and import it in FreeCAD 0.19 where the assembly works fine then.

feichtinger, 2021-09-27:
This issue is still present with A2plus V0.4.54b. Especially when I try to use imported (STEP) models (e.g. PCB with components) A2plus often has a hard time. When I try to add the attached part to an assembly I get the following message:

`13:20:44  Running the Python command 'a2p_ImportPart' failed:
Traceback (most recent call last):
  File "/home/user/.FreeCAD/Mod/A2plus/a2p_importpart.py", line 736, in Activated
    importedObject = importPartFromFile(doc, filename)
  File "/home/user/.FreeCAD/Mod/A2plus/a2p_importpart.py", line 372, in importPartFromFile
    topoMapper.createTopoNames()
  File "/home/user/.FreeCAD/Mod/A2plus/a2p_topomapper.py", line 620, in createTopoNames
    needDiffuseExtension = ( len(ob.ViewObject.DiffuseColor) < len(ob.Shape.Faces) )

'Gui.ViewProviderDocumentObject' object has no attribute 'DiffuseColor'`

[BGA.zip](https://github.com/kbwbe/A2plus/files/7235826/BGA.zip)

Thanks for help.

OS: Ubuntu 18.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.
Build type: Release
Branch: unknown
Hash: 582c26ea8b5f9d85957a6cb1bf798839fa2ce94c
Python version: 3.6.9
Qt version: 5.9.5
Coin version: 4.0.0a
OCC version: 7.5.2
Locale: German/Germany (de_DE)

for-public-contact, 2021-12-21:
I have attached an **object that may help you resolve this issue**;
[Modelling Error - object has no attribute DiffuseColor.zip](https://github.com/kbwbe/A2plus/files/7755001/Modelling.Error.-.object.has.no.attribute.DiffuseColor.zip)

If it's importing as-is - it produces an error.
However, if you pull out all 3 items from the simple group 'Roller block' outside and delete this group - the import will be successful.