kbwbe/A2plus #301: [FC 0.19pre] Constraints not in container

maximbendiks, 2019-10-14:
When creating constraints with A2+ v0.4.27a in FC0.19 they do not group into all the containers of all parts (see in the picture, where all constraints of the first part are in not in the container but all constraints of the second part are in the corresponding container). 
[image]
Closing and reopening the file resolves the issue until you create new constraints. Those are again out of the containers.
[constraints.zip](https://github.com/kbwbe/A2plus/files/3723713/constraints.zip)

In FC0.18 it looks ok.

OS: Manjaro Linux (dwm/dwm)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18513 (Git)
Build type: Release
Branch: master
Hash: fd352c14f13ecdef7b8d4005c987d2f449c69430
Python version: 3.7.4
Qt version: 5.13.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

kbwbe, 2019-10-18:
I also registered this effect. Closing and reopening file helped and the constraints were correctly sorted beneath parts again. I am wondering whether it has to do with A2p or it is a bug in FC0.19-pre.

donovaly, 2020-01-17:
The same issue was in TechDraw:
https://forum.freecadweb.org/viewtopic.php?f=35&t=42413
It was fixed by this commit:
https://github.com/FreeCAD/FreeCAD/commit/8b2e318f15

So maybe a similar fix needs to be applied to A2plus?

kbwbe, 2020-01-18:
Thank you, i will try this.

closed 2020-01-18 by kbwbe

kbwbe, 2020-01-18:
Hi donovaly, this was a good hint. The proposed fix worked. Thank you!

PR #337 (2020-01-18): Temp
fixes #336, fixes #301