kbwbe/A2plus #303: Crashes while importing part and pressing ESC

maximbendiks, 2019-10-14:
When you press ESC while importing a file and you are in that state, where the part is following the mouse for placement, a segfault happens:
```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3bfb0) [0x7f92bff54fb0]
#1  0x7f92c206c247 in SoAction::traverse(SoNode*) from /usr/lib/libCoin.so.80+0x27
#2  0x7f92c2228bbb in SoChildList::traverse(SoAction*, int, int) from /usr/lib/libCoin.so.80+0xeb
#3  0x7f92c2276bd8 in SoGroup::doAction(SoAction*) from /usr/lib/libCoin.so.80+0x58
#4  0x7f92c2291daa in SoSeparator::doAction(SoAction*) from /usr/lib/libCoin.so.80+0x2a
#5  0x7f92c3e25770 in Gui::SoFCUnifiedSelection::handleEvent(SoHandleEventAction*) from /home/maxim/workspace/freecad_19/lib/libFreeCADGui.so+0x90
#6  0x7f92c206c28a in SoAction::traverse(SoNode*) from /usr/lib/libCoin.so.80+0x6a
#7  0x7f92c2228bbb in SoChildList::traverse(SoAction*, int, int) from /usr/lib/libCoin.so.80+0xeb
#8  0x7f92c2276bd8 in SoGroup::doAction(SoAction*) from /usr/lib/libCoin.so.80+0x58
#9  0x7f92c2291daa in SoSeparator::doAction(SoAction*) from /usr/lib/libCoin.so.80+0x2a
#10  0x7f92c206c28a in SoAction::traverse(SoNode*) from /usr/lib/libCoin.so.80+0x6a
#11  0x7f92c2075f2c in SoHandleEventAction::beginTraversal(SoNode*) from /usr/lib/libCoin.so.80+0xac
#12  0x7f92c206d0d2 in SoAction::apply(SoNode*) from /usr/lib/libCoin.so.80+0x442
#13  0x7f92c2242462 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /usr/lib/libCoin.so.80+0x72
#14  0x7f92c3ea12ee in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /home/maxim/workspace/freecad_19/lib/libFreeCADGui.so+0x1e
#15  0x7f92c3ea9567 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /home/maxim/workspace/freecad_19/lib/libFreeCADGui.so+0x57
#16  0x7f92c3edc00f in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /home/maxim/workspace/freecad_19/lib/libFreeCADGui.so+0xaf
#17  0x7f92c3e9a311 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /home/maxim/workspace/freecad_19/lib/libFreeCADGui.so+0xd1
#18  0x7f92c09d06d3 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x93
#19  0x7f92c138c4e4 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x74
#20  0x7f92c1396176 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x696
#21  0x7f92c3c0c748 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/maxim/workspace/freecad_19/lib/libFreeCADGui.so+0x88
#22  0x7f92c09d09c2 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x182
#23  /usr/lib/libQt5Widgets.so.5(+0x1bc2fe) [0x7f92c13ef2fe]
#24  0x7f92c138c4f5 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x85
#25  0x7f92c1395e11 in QApplication::notify(QObject*, QEvent*) from /usr/lib/libQt5Widgets.so.5+0x331
#26  0x7f92c3c0c748 in Gui::GUIApplication::notify(QObject*, QEvent*) from /home/maxim/workspace/freecad_19/lib/libFreeCADGui.so+0x88
#27  0x7f92c09d09c2 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /usr/lib/libQt5Core.so.5+0x182
#28  0x7f92c0d6b03d in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /usr/lib/libQt5Gui.so.5+0x16d
#29  0x7f92c0d7047f in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) from /usr/lib/libQt5Gui.so.5+0x25f
#30  0x7f92c0d4aafc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Gui.so.5+0xbc
#31  /usr/lib/libQt5XcbQpa.so.5(+0x6c12c) [0x7f92bb77412c]
#32  /usr/lib/libglib-2.0.so.0(g_main_context_dispatch+0x27e) [0x7f92bd3c73fe]
#33  /usr/lib/libglib-2.0.so.0(+0x6c211) [0x7f92bd3c9211]
#34  /usr/lib/libglib-2.0.so.0(g_main_context_iteration+0x31) [0x7f92bd3c9251]
#35  0x7f92c0a28a03 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x63
#36  0x7f92c09cf4ec in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /usr/lib/libQt5Core.so.5+0x12c
#37  0x7f92c09d7326 in QCoreApplication::exec() from /usr/lib/libQt5Core.so.5+0x96
#38  0x7f92c3b84c39 in Gui::Application::runApplication() from /home/maxim/workspace/freecad_19/lib/libFreeCADGui.so+0x1459
#39  ./bin/FreeCAD(main+0x6f6) [0x561d8c18f7b6]
#40  /usr/lib/libc.so.6(__libc_start_main+0xf3) [0x7f92bff40153]
#41  ./bin/FreeCAD(_start+0x2e) [0x561d8c18fabe]
```
Happens in FC0.19 and FC0.18.
OS: Manjaro Linux (dwm/dwm)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.18513 (Git)
Build type: Release
Branch: master
Hash: fd352c14f13ecdef7b8d4005c987d2f449c69430
Python version: 3.7.4
Qt version: 5.13.1
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: German/Germany (de_DE)

kbwbe, 2019-10-18:
I will check this...

kbwbe, 2019-10-26:
The crash's should be fixed now. Please update your A2plus WB to version v0.4.28

closed 2019-10-26 by kbwbe