kbwbe/A2plus #311: Can freely move parts even under constraints

KoltesDigital, 2019-12-08:
Hi,

Given two objects, fully constraints together, and AutoSolve is on. Selecting one of them and pressing "Move the selected part", it can be freely moved. After pressing "Solves constraints" it goes back in place.

Is it intended? As the AutoSolve tooltip states, solving is only done after adding a new constraints. I'd have expected that it'd be done when moving as well, in order to see how the assembly works. While it's also useful to temporarily break the assembly, this could be achieved by switching off AutoSolve.

kbwbe, 2019-12-08:
It is the intended behaviour. Sometimes you have to move a part away because it is hiding the geometry with which you want to work on next.

KoltesDigital, 2019-12-08:
Yes I mentioned that in my last sentence. This behavior can be kept when AutoSolve is switched off. I believe it'd be great to have both behaviors, and to choose between them with the AutoSolve button.

kbwbe, 2019-12-09:
For me, i am mostly working with AutoSolve on and i need to move parts without solving.
I think it would be better to have a second, different move button for this, where movement triggers solving. But this will be surely slow, as the solver is implemented in pure python.

KoltesDigital, 2019-12-10:
OK then it makes sense to have another button.

I don't know how solvers are implemented. I thought it would be able to give a list like "this part has TX and RY DOF" so it'd be enough to run it once to know how parts can move without running it again. But even if it's not the case, I think it's worth a try to see whether it's too slow.

kbwbe, 2019-12-10:
I have added this to my todo list. It will a little bit time, as i am busy with other things.

KoltesDigital, 2019-12-11:
Sure, it's not a priority feature. Thanks!

kbwbe, 2019-12-22:
See version v0.4.33
There is a new button "movePartsUnderConstraints". Now you can select a part by mouse and move it around. It is moving according to the constraints of the model.
Please follow the instructions which are shown in the tooltip.

closed 2019-12-22 by kbwbe

KoltesDigital, 2019-12-22:
Awesome, thanks!