kbwbe/A2plus #313: Issue when importing a file with the same name as the file being edited.

leoheck, 2019-12-12:
Hi, I am using A2Plus for a long time it is my main workbench after the part design. My current version is the master version (0.4.30). Using it on Freecad 0.19 (daily builds)

I am seeing this issue for such a long time too.
When I am importing a file with the same name as the current file being edited, it is adding a copy of the file being edited. 

I usually store my 3d printing projects in folders, and inside of each folder, I have many FCStd files that are a single printable part. Then I have in each folder a file called Assembly.

Then sometimes I want to compare the old assembly with this new assembly being created to see how much my part has changed over time.

kbwbe, 2019-12-12:
HI,
isn't it better you create a new "asm-comparison.fcstd" file and import your various assembly states to this new file, in order to compare them ?

The recent behaviour of A2plus cannot easily be changed. The main criteria for import of a file is it's name. A2plus always tries to take the import information from files which are already open, in order to save time.

leoheck, 2019-12-12:
I see. Well, I believe that comparing filenames is not a good idea et al. But one thing I can think to improve this is to use the file path (relative or not) to do the same job.

For example, instead of comparing things like ```assembly.FCStd``` you can use bigger file names like ```/absolute/path/to/file.FCStd``` or even better ```../path/relative/to/file.FCStd```

I believe this is a small fix to overcome the issue.

What do you think?

ramezquitao, 2021-11-12:
For me this is an issue also. I have some assembly models that are used as sub-assemblies in the final model. All this assemblies and sub-assemblies live in different sub-folders, but may have files with names in common. Because there are files with the same name in the different sub-assemblies, an issue similar to the one described is happening from time to time to me. 
This is not something critical, but it can be very annoying.
By the way. Thanks for developing this workbench. It is very useful.

luzpaz, 2023-05-15:
@kbwbe please weigh-in when you have a moment