kbwbe/A2plus #316: How to use the constraints?

leoheck, 2019-12-15:
I love the A2Plus workbench but I never could use the constraints.

I always get this message. No matter what I do. And I don't know what to do. Do you have any tutorials on how to use them?
[image]

I am using Freecad 0.19 right now and A2Plus v 0.4.29a (from the master)

leoheck, 2019-12-15:
I am adding parts with the `Shift+a` shortcut then trying to constraint them. I am starting to think it is wrong because I saw this `Convert a part...` tool which has this yellow-red icon. However, I am not able to constrain the objects after converting them with this tool.

kbwbe, 2019-12-16:
At least one Part has to have the property "fixed Position" = True. That's all.

closed 2019-12-17 by kbwbe