kbwbe/A2plus #317: Freecad crashes when I click "edit an imported part"

svancina, 2019-12-15:
I just started working with A2plus. I opened a new project and used the A2plus workbench to add a part form an external file (a model of a MakerBeamXL extrusion). After importing that part I click "edit an imported part" from the toolbar. At this point Freecad immediately crashes to the desktop. Happens every time I try to edit the imported part. I'm new to bug reporting, so let me know what other info would be helpful!

My system is:
- Kubuntu 19.10, 64-bit
- Linux kernel 5.3.0-24-generic
- Freecad 0.18 rev 16146
- A2plus V0.4.31
- AMD FX-8350 CPU
- 16GB RAM

kbwbe, 2019-12-17:
No issues here. You can start freecad from a console with the command "freecad -l".
With the -l parameter, freecad creates a crash dump within the FC folder. Perhaps it is possible to see there why your system is crashing.

svancina, 2019-12-18:
Here's the terminal output I get when the program crashes after running with "freecad -l"

[FreeCAD_Crash_Log_2019-12-18_2029UTC.txt](https://github.com/kbwbe/A2plus/files/3980392/FreeCAD_Crash_Log_2019-12-18_2029UTC.txt)

Interestingly I tried installing FreeCAD from the snap store (latest in the store is v0.18 rev 16093), and that version does not crash when I try to edit the assembly part. The program also starts up much faster and seems more responsive. The version I had previously was the appimage downloaded from the FreeCAD website.

kbwbe, 2019-12-22:
I think it is a problem of the basic FreeCAD version and installation and not a problem of A2plus. Otherwise there would have been much more response from many people here and in the FC Forum.
Anyway, thank you for reporting.

closed 2019-12-22 by kbwbe