kbwbe/A2plus #341: Updating A2plus parts by script

leoheck, 2020-01-20:
Hi, sorry to add this here but I also posted it in Freecad Forum, and it seems that no one is available to help this time.

https://forum.freecadweb.org/viewtopic.php?f=22&t=42532

Can you show me how to update parts from a script? Is it possible? I am not using GUI.
In the python console I saw that the  update icons on toolbar generate this:

```
Gui.activateWorkbench("A2plusWorkbench")
Gui.runCommand('a2p_updateImportedParts',0)
```

However, I don't have this Gui object from the script without Gui. Any idea, reference or help would be nice. Thank you `kbwbe`

kbwbe, 2020-01-20:
Is is possible to do something like here: [https://forum.freecadweb.org/viewtopic.php?f=22&t=39470](url)
With this:
`        import FreeCAD, FreeCADGui
        FreeCADGui.showMainWindow()
        mw = FreeCADGui.getMainWindow()
        mw.hide()
`
you can run FC in headless mode and have Gui available.

leoheck, 2020-01-20:
I tried that a little bit, I don't like the approach, but it opens the Gui, then suddenly hides it and after that my script breaks.

I saw that I was not importing A2Plus scripts. So I could import `import A2Plus` by appending the python path to `~/.FreeCAD/Mod/`. I think now I can use the A2Plus functions directly but I couldn't verify this yet.

leoheck, 2020-03-18:
I posted my question [here on the Freecad forum](https://forum.freecadweb.org/viewtopic.php?f=22&t=44264), and I got some insights on how to solve my issue. However, this plugin is a little bit complex for me to understand. Do you have any insight for me to expose this `a2p_updateImportedParts` method so I can use it from a script and without the GUI?

luzpaz, 2023-05-15:
@kbwbe ping